jvpernis/esp32-ps3 #1: can't find ESP_BT_CONNECTABLE variable

blooop, 2019-07-06:
I've installed the esp idf framework and ran the hello world example.

I copied the code snipped on the readme into the hello world main.c and put your ps3 library and the  esp-idf/components/bt into the components folder in the hello world example.  For good measure I also copied your ps3 folder into the components directory in the esp-idf/components folder.  Is that what you mean by install globally?

However when I run make flash I get this error.   It looks like I have not included  esp_gap_bt_api.h properly, but I'm new to this and must have done it incorrectly.  Do you have any tips?  Thanks

make output:

```Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from D:/msys32/home/ags/esp/esp-idf/requirements.txt are satisfied.
CC build/ps3/ps3_spp.o
D:/msys32/home/ags/esp/hello_world/components/ps3/ps3_spp.c: In function 'ps3_spp_callback':
D:/msys32/home/ags/esp/hello_world/components/ps3/ps3_spp.c:95:34: error: 'ESP_BT_CONNECTABLE' undeclared (first use in this function)
         esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);
                                  ^
D:/msys32/home/ags/esp/hello_world/components/ps3/ps3_spp.c:95:34: note: each undeclared identifier is reported only once for each function it appears in
D:/msys32/home/ags/esp/hello_world/components/ps3/ps3_spp.c:95:54: error: 'ESP_BT_NON_DISCOVERABLE' undeclared (first use in this function)
         esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);
                                                      ^
D:/msys32/home/ags/esp/hello_world/components/ps3/ps3_spp.c:95:9: error: too many arguments to function 'esp_bt_gap_set_scan_mode'
         esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);
         ^
In file included from D:/msys32/home/ags/esp/hello_world/components/ps3/ps3_spp.c:13:0:
D:/msys32/home/ags/esp/esp-idf/components/bt/bluedroid/api/include/api/esp_gap_bt_api.h:361:11: note: declared here
 esp_err_t esp_bt_gap_set_scan_mode(esp_bt_scan_mode_t mode);
           ^
make[1]: *** [/home/ags/esp/esp-idf/make/component_wrapper.mk:286: ps3_spp.o] Error 1
make: *** [D:/msys32/home/ags/esp/esp-idf/make/project.mk:530: component-ps3-build] Error 2
```

jvpernis, 2019-07-08:
> error: too many arguments 

This tells me you are not on the latest version of the esp-idf library.

As of commit [d9ce0bb](https://github.com/espressif/esp-idf/commit/d9ce0bb705b155c2f51aabc929026888a3d7a7e8), on Apr 9, 2018, the function `esp_bt_gap_set_scan_mode` was modified to split the single scan-mode argument into two arguments: connectable-mode and discoverable mode.

It looks like you installed my library properly, so you could try to update your esp-idf framework. Please let me know if this helps!

blooop, 2019-07-08:
Thanks,  I worked out that the tutorial was out of date and linking to an old version of esp-idf.  I updated it to v3.3 yesterday and still got the same errors.  I'm guessing I must have messed something up so my plan for  this afternoon was to wipe everything and start again from scratch.  I'll let you know if that fixes it.

jvpernis, 2019-07-08:
Oh, I just realized the commit I referenced hasn't made it to a stable version yet. You can try installing the latest version from the master branch. I haven't yet tried compiling it, but it should work. 

I will try to fabricate a solution somewhere in this week

blooop, 2019-07-08:
I've updated esp-idf to master and added your library to the esp-idf/components folder.

It comes up with this error:
```
make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from D:/msys32/home/ags/esp/esp-idf/requirements.txt are satisfied.

Project is not inside a git repository, will not use 'git describe' to determine PROJECT_VER.
App "hello-world" version: 1
CC build/ps3/ps3_l2cap.o
D:/msys32/home/ags/esp/esp-idf/components/ps3/ps3_l2cap.c:11:10: fatal error: stack/gap_api.h: No such file or directory
 #include "stack/gap_api.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [/home/ags/esp/esp-idf/make/component_wrapper.mk:292: ps3_l2cap.o] Error 1
make: *** [D:/msys32/home/ags/esp/esp-idf/make/project.mk:584: component-ps3-build] Error 2
```

Which commit of esp-idf are you on?  If I just check that one out I assume everything should work without you needing to make a fix. 

Thanks

blooop, 2019-07-08:
I messed about a bit trying to fix the include errors.  I added:

```
                                $(IDF_PATH)/components/bt/host/bluedroid/stack/include        \
                                $(IDF_PATH)/components/bt/host/bluedroid/common/include       \
                                $(IDF_PATH)/components/bt/common/include    
```
to the component.mk which fixed all the include errors, but now there is a non include related error:

```
CC build/ps3/ps3.o
D:/msys32/home/ags/esp/esp-idf/components/ps3/ps3.c: In function 'ps3Cmd':
D:/msys32/home/ags/esp/esp-idf/components/ps3/ps3.c:108:5: error: 'memcpy' forming offset [51, 52] is out of the bounds [0, 50] of object 'hid_cmd' with type 'hid_cmd_t' {aka 'struct <anonymous>'} [-Werror=array-bounds]
     memcpy( hid_cmd.data, hid_cmd_payload, len);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:/msys32/home/ags/esp/esp-idf/components/ps3/ps3.c:86:15: note: 'hid_cmd' declared here
     hid_cmd_t hid_cmd = { .data = {0} };
               ^~~~~~~
cc1.exe: some warnings being treated as errors
make[1]: *** [/home/ags/esp/esp-idf/make/component_wrapper.mk:292: ps3.o] Error 1
make: *** [D:/msys32/home/ags/esp/esp-idf/make/project.mk:584: component-ps3-build] Error 2
make: *** Waiting for unfinished jobs....

```

jvpernis, 2019-07-09:
I just checked and my system is currently at [547790](https://github.com/espressif/esp-idf/commit/76da10a665c228b81225130ca2060d04c1ee000a). I'm sorry for the inconvenience!

blooop, 2019-07-09:
Thanks, that works now.

For anyone else reading this, these are the commands I used to clone esp-idf at the right commit:
```
#clone without checkout because git complains otherwise during the checkout
git clone -n https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout 76da10a665c228b81225130ca2060d04c1ee000a
git submodule update -j 4 --init --recursive

./tools/windows/windows_install_prerequisites.sh
python -m pip install --user -r $IDF_PATH/requirements.txt
```

closed 2019-07-09 by blooop

ItsMeAhmed, 2019-07-12:
> to the component.mk which fixed all the include errors, but now there is a non include related error:
>
> ```
> CC build/ps3/ps3.o
> D:/msys32/home/ags/esp/esp-idf/components/ps3/ps3.c: In function [...]
Hi, I faced the same problem at first, fixed it by editing the following line:
```
memcpy( hid_cmd.data, hid_cmd_payload, len);
```
to
```
memcpy( hid_cmd.data, hid_cmd_payload, sizeof(hid_cmd.data));
```

jvpernis, 2019-07-13:
Hi!

I wanted to let you know that I added a config menu item in which you can select which ESP-IDF revision you'd like to be compatible with. With this, it should be possible again to compile from the latest master revision.

By default, it is set to be compatible with the latest stable release (v3.3 at the time of writing).

> to the component.mk which fixed all the include errors, but now there is a non include related error:

> Hi, I faced the same problem at first, fixed it by editing the following line

I also fixed this error, which, for some strange reason, didn't happen with my setup when I upgraded to the latest revision.

blooop, 2019-07-16:
perfect, thanks.

Omegaki113r, 2020-05-13:
can you explain what is the purpose of, ESP_BT_LIMITED_DISCOVERABLE and ESP_BT_NON_DISCOVERABLE.

I can assume that ESP_BT_NON_DISCOVERABLE means scanning from android will not show the ESP32 device. if so how can i connect to it if that ESP32 is NON_DISCOVERABLE?