jvpernis/esp32-ps3 #2: Minimal example not connecting

blooop, 2019-07-09:
Hi, sorry to be spamming you with issues and thanks for you help with the install. 

I used the sixaxis pair tool to set the ps3 BT master to the value that is printed during flashing.

```
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: b4:e6:2d:ee:7b:fd
Uploading stub...
```
The code in the main file is modified from the hello_world example:

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "ps3.h"
#include "nvs_flash.h"  //included to stop some nvs errors

void controller_event_cb(ps3_t ps3, ps3_event_t event)
{
    if (ps3.button.triangle)
        printf("Currently pressing the trangle button");
}

void app_main()
{
    printf("Before setup\n");
    ps3SetEventCallback(controller_event_cb);

    //ADDED THIS TO STOP ERROR MESSAGES COMPLAING ABOUT NVS FLASH INIT
    nvs_flash_init();

    ps3Init();
    while (!ps3IsConnected())
        ;

    printf("After setup\n");
}
```

There were errors to do with nvs_flash, but I fixed that by calling nvs_flash_init().  However, the code still seems not to connect.   This is the output of make monitor which never reaches printf("After setup\n"):

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6352
ho 0 tail 12 room 4
load:0x40078000,len:11308
load:0x40080400,len:6680
entry 0x40080760
I (30) boot: ESP-IDF v4.0-dev-207-g76da10a66 2nd stage bootloader
I (30) boot: compile time 18:45:07
I (39) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x23354 (144212) map
I (145) esp_image: segment 1: paddr=0x0003337c vaddr=0x3ffbdb60 size=0x02a80 ( 10880) load
I (150) esp_image: segment 2: paddr=0x00035e04 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at D:/msys32/home/ags/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (152) esp_image: segment 3: paddr=0x0003620c vaddr=0x40080400 size=0x09e04 ( 40452) load
I (178) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x74a88 (477832) map
0x400d0018: _flash_cache_start at ??:?

I (345) esp_image: segment 5: paddr=0x000b4aa8 vaddr=0x4008a204 size=0x080b4 ( 32948) load
0x4008a204: r_lld_pdu_rx_handler at ??:?

I (370) boot: Loaded app from partition at offset 0x10000
I (370) boot: Disabling RNG early entropy source...
I (370) cpu_start: Pro cpu up.
I (374) cpu_start: Application information:
I (379) cpu_start: Project name:     hello-world
I (384) cpu_start: App version:      1
I (388) cpu_start: Compile time:     Jul  9 2019 18:44:07
I (394) cpu_start: ELF file SHA256:  b7c0231e2b184b06...
I (400) cpu_start: ESP-IDF:          v4.0-dev-207-g76da10a66
I (407) cpu_start: Starting app cpu, entry point is 0x400810a0
0x400810a0: call_start_cpu1 at D:/msys32/home/ags/esp/esp-idf/components/esp32/cpu_start.c:267

I (0) cpu_start: App cpu up.
I (417) heap_init: Initializing. RAM available for dynamic allocation:
I (424) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (430) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (436) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (442) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (448) heap_init: At 3FFCCF78 len 00013088 (76 KiB): DRAM
I (455) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (461) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (467) heap_init: At 400922B8 len 0000DD48 (55 KiB): IRAM
I (474) cpu_start: Pro cpu start user code
I (156) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Before setup
I (218) BTDM_INIT: BT controller compile version [80351af]
I (218) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (308) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0
I (918) PS3_SPP: ESP_SPP_INIT_EVT
I (938) PS3_L2CAP: [ps3_l2cap_init_service] Service PS3-HIDC Initialized
I (938) PS3_L2CAP: [ps3_l2cap_init_service] Service PS3-HIDI Initialized
E (5918) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5918) task_wdt:  - IDLE0 (CPU 0)
E (5918) task_wdt: Tasks currently running:
E (5918) task_wdt: CPU 0: main
E (5918) task_wdt: CPU 1: IDLE1

```

I also tried to set the mac address but that gets the same results:

```
static const char macStr[] = "b4:e6:2d:ee:7b:fd";
    unsigned int iMac[6];
    uint8_t mac[6];
    sscanf(macStr, "%x:%x:%x:%x:%x:%x", &iMac[0], &iMac[1], &iMac[2], &iMac[3], &iMac[4], &iMac[5]);
    int i;
    for (i = 0; i < 6; i++)
        mac[i] = (unsigned char)iMac[i];
    esp_base_mac_addr_set(mac);
```

Thanks

jvpernis, 2019-07-09:
Hi, don't worry, I'm actually quite happy to know someone's working with the library!

This one should be quite simple, as I see your logs stating the Task Watchdog has been triggered. This is some kind of timer that limits how long a task is allowed to run. If a task exceeds this limit, the watchdog thinks the software got stuck, and will (usually) reset the software in hopes it doesn't happen again.

Now this is actually a fault in my example code:

```c
while (!ps3IsConnected());
```

This code snippet literally blocks the task from continuing until the PS3 controller has been connected.

To remedy this behavior, you could either:

- [Disable the Task Watchdog ](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-esp-task-wdt) altogether 
- Or call the [vTaskDelay](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/freertos.html#_CPPv410vTaskDelayK10TickType_t) function within the while loop, as this resets the watchdog timer:
```c
#include "freertos/task.h"
```

```c
while (!ps3IsConnected()){
    vTaskDelay(10 / portTICK_PERIOD_MS);
}
```
Again, I'm not able to compile this at the moment to test this for you, but I will try to update the example code this week.

blooop, 2019-07-10:
Thanks, that fixes the watchdog, but the code still doesn't connect.  Did I set up the Bluetooth address correctly?

jvpernis, 2019-07-10:
It looks like your code should work just fine. You could try to set the MAC address even before calling `nvs_flash_init()`.

I also vaguely remember having to reverse the MAC address written to the PS3 controller, so you could try that too:

- MAC on ESP32: b4:e6:2d:ee:7b:fd
- MAC on PS3: fd:7b:ee:2d:e6:b4

If that's the case I will update the README accordingly.

Also note, whenever you connect the PS3 controller with your PS3 to charge, it will also overwrite the stored MAC address.

blooop, 2019-07-10:
That has not worked I'm afraid.  Is there any debug info I can print to see what is going wrong?

ItsMeAhmed, 2019-07-11:
Hi, I am facing the same issue here, tried all mentioned solutions but the controller is still not connecting.

jvpernis, 2019-07-11:
I think I have just found the catch to this issue, which is something I completely forgot I ever configured.

At the start of the program, we are assigning a new **_base_** MAC address. From the [IDF documentation about it](https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/system/system.html?highlight=esp_base_mac_addr_set#mac-address) we can learn that this base address is taken to generate a different MAC address for WiFi, Bluetooth and Ethernet.

To get the bluetooth MAC address, you have to add 2 to the last byte of your base MAC address:
* MAC on ESP32: b4:e6:2d:ee:7b:fd 
* MAC on PS3: b4:e6:2d:ee:7b:ff 

Please let me know if this works! I'll then add this info to the README as well.

ItsMeAhmed, 2019-07-11:
It finally worked!! ,, excellent job @jvpernis 
[image]

ItsMeAhmed, 2019-07-11:
here is the code i used ,,,
`
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "ps3.h"
#include "nvs_flash.h" 

void controller_event_cb( ps3_t ps3, ps3_event_t event )
{

    if ( event.button_down.cross )
        printf("The user started pressing the X button\n");

    if ( event.button_up.cross )
        printf("The user released the X button\n");
}

void app_main()
{
    printf("Hello world!\n");
    uint8_t new_mac[8] = {0x30,0xae,0xa4,0xbf,0x5b,0x7e}; //the address on the controller is 30:AE:A4:BF:5B:80
    esp_base_mac_addr_set(new_mac);
    nvs_flash_init();

    ps3SetEventCallback(controller_event_cb);
    ps3Init();
    while (!ps3IsConnected()){
    vTaskDelay(10 / portTICK_PERIOD_MS);
	}
}

`

ItsMeAhmed, 2019-07-11:
> It works!
> 
> Thanks for all the help!
> 
> For future reference, if my last two characters were :ff would it roll over to f1?

maybe it'll carry over to the next digit,, 30:AE:A4:BF:5B:ff + 2 = 30:AE:A4:BF:5C:01 ?

blooop, 2019-07-11:
Ah yes of course, I forgot to increment in the binary representation. 

For reference my code is almost identical except I don't set the base mac address and just set the controller to the hardcoded mac address +2.

jvpernis, 2019-07-11:
> It works!

Nice!

> For future reference, if my last two characters were :ff would it roll over to f1?

That is actually a really interesting question! According to [the source code](https://github.com/espressif/esp-idf/blob/76da10a665c228b81225130ca2060d04c1ee000a/components/esp32/system_api.c#L185), it appears that just the last byte (i.e. the last two hexadecimal characters) will be affected, and it would simply rollover to 01.

I  just tested it and indeed, I could connect with the following MAC:
* MAC on ESP32: 01:02:03:04:05:ff
* MAC on PS3: 01:02:03:04:05:01

jvpernis, 2019-07-13:
Hi!

I wanted to let you know that I updated the code snipped in the README to include the `vTaskDelay` function in order to prevent the Task Watchdog from triggering. 

I also created a function `ps3SetBluetoothMacAddress`, which can be used to set the Bluetooth MAC address identical as the one stored in the PS3 controller. You no longer have to apply a +2 operation to the latest byte, as the new function does this for you.

blooop, 2019-07-16:
great, thanks. I was going to write that function :)

jvpernis, 2019-07-23:
I'll be closing this issue, as I believe the problems described have been resolved. Please feel free to reopen the issue if you disagree with this conclusion!

closed 2019-07-23 by jvpernis