jvpernis/esp32-ps3 #3: Arduino Library?

Sniperblue, 2019-07-22:
Hi, is it scheduled to release also a modify Library based on this Espressif one which works with Arduino IDE. Up to now I think there is nothing around like this in the net. Unfortunaley I'am not such a Bleutooth specialist to do this by myself :-(

jvpernis, 2019-07-23:
Hi, thanks for your interest in using this library!

Releasing an Arduino library actually hasn't crossed my mind yet. I'd like to look into how feasible porting this library would be.

However, the API exposed by Espressif's Arduino implementation is vastly diffrent from the ESP-IDF framework. Next to that, I'd probably need to rewrite this library's API as well in order to conform to the Arduino API style guide. Because of these differences, it would be most sensible to create a new repository if I ever come around to porting it to Arduino.

Don't expect it anytime soon though, but I'll try to make some time to create an Arduino library.

I'll keep this issue open so you'll get notified whenever I get around to creating the Arduino library :)

navar00, 2019-07-23:
Same here, it would be great to have a port into arduino. There is nothing off the self so far and it is highly demanded.....

Sniperblue, 2019-07-25:
This would be great! I remain in hopeful expectation...

jvpernis, 2019-08-01:
Good news! 

As it turns out, adding support for the Arduino IDE wasn't so hard after all. I didn't even have to create another repository for it, as it works quite nicely together with the ESP-IDF!

### **Now I need _your_ help!**

Before releasing it as an official Arduino library, I want to polish things up a little. In the mean time, though, I could use your help in testing this library, to see if it suits your needs. Does it work properly? Does it miss any features? Would you prefer other naming schemes for the functions and variables? Maybe you can think of a better name for the library? Let me know what you think.

**Installing the ESP32 boards**

In case you haven't yet, you can add the ESP32 boards to your Arduino IDE by adding them to the Boards Manager: Open `File -> Preferences`, and paste the following URL in the `Additional Boards Manager URLs` field:

- `https://dl.espressif.com/dl/package_esp32_index.json`

Open the Boards Manager with `Tools -> Board: "xxx" -> Boards Manager`, look for `esp32` (probably the last one in the list), and click `install`.

Finally, select the ESP32 board you have with `Tools -> Board: "xxx"` under the section `ESP32 Arduino` (I have `ESP32 Dev Module` selected).

**Installing the Ps3Controller library**

You can download the Arduino library from the development branch: [esp32-ps3-develop.zip](https://codeload.github.com/jvpernis/esp32-ps3/zip/develop).

Then, in the Arduino IDE, you can import the library with `Sketch -> Include Library -> Add .ZIP Library...`, and selecting the downloaded `esp32-ps3-develop.zip` file.

**Running an example sketch**

The example sketches should be immediately available in the IDE. Try to connect to the PS3 controller first with the Ps3Connect sketch: `File -> Examples -> Ps3Controller -> Ps3Connect`.

Don't forget to properly pair the PS3 controller as described [here](https://github.com/jvpernis/esp32-ps3#pairing-the-ps3-controller).

Hope this helps!

Sniperblue, 2019-08-06:
When I try to complie you sketch I get the following error Message:

_In file included from C:\Users\Sniperblue\Documents\Arduino\libraries\esp32-ps3-develop\src\ps3_spp.c:6:0:

C:\Users\Sniperblue\Documents\Arduino\libraries\esp32-ps3-develop\src\include/ps3_int.h:29:2: error: #error "The selected Bluetooth controller mode is not supported by the ESP32-PS3 module"

 #error "The selected Bluetooth controller mode is not supported by the ESP32-PS3 module"_

I use Arduino Version 1.8.9 with ESP Version 1.0.1 installed and try to compile withESP32 Dev Module and DOIT ESP32 Dev V1 selected. The ESP32 says this ESP32 SDK Version: v3.2-dev-39-gaaf12390

Hope this helps to get an idea about the root cause..!?

jvpernis, 2019-08-06:
Thanks for trying it out! Unfortunately, I am unable to replicate your problem. I just tried to execute the steps I described above on a fresh Windows install, using Arduino Version 1.8.9 with ESP Version 1.0.1 installed, just like you described. Even with the DOIT ESP32 Dev V1 board selected it compiles just fine for me.

Could you try to restart the Arduino IDE, or even your computer, and try again? 

Have you changed any configuration settings using `make menuconfig`? 

Could you share how you've set up your Arduino environment?

Sniperblue, 2019-08-06:
I can't rember that I changed any configuration settings using make menuconfig.
 
What do you mean with share how you've set up your Arduino environment, preferences.txt file from the ...AppData\Local\Arduino15 folder or something like that?

jvpernis, 2019-08-06:
With sharing your setup I meant how you installed the Arduino IDE and the ESP SDK. But yes, seeing your `...AppData\Local\arduino15\preferences.txt` file could help, and maybe even your `...AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.1\tools\sdk\sdkconfig` file if you can.

The error you get means a misconfiguration in the sdkconfig file, where BTDM_CONTROLLER_MODE is set to BLE, while it should be set to any of the other options which enables the Classic Bluetooth ('Both' or 'BR/EDR Only'). The default config shipped with the ESP32 boards should be adequate in this regard, which is why I think it is really strange this is happening to you!

Sniperblue, 2019-08-06:
ESP32 was installed in a standart was as described here:
https://randomnerdtutorials.com/installing-the-esp32-board-in-arduino-ide-windows-instructions/

Here are the BT extract form the sdkconfig file:
```
#
# Bluetooth
#
CONFIG_BT_ENABLED=y

#
# Bluetooth controller
#
CONFIG_BTDM_CONTROLLER_MODE_BLE_ONLY=
CONFIG_BTDM_CONTROLLER_MODE_BR_EDR_ONLY=
CONFIG_BTDM_CONTROLLER_MODE_BTDM=y
CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN=3
CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN=2
CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN=0
CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN_EFF=3
CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN_EFF=2
CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN_EFF=0
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_0=y
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_1=
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE=0
CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI=y
CONFIG_BTDM_CONTROLLER_HCI_MODE_UART_H4=
```
Here are the sdkconfig:
[sdkconfig.txt](https://github.com/jvpernis/esp32-ps3/files/3473950/sdkconfig.txt)

Here are the preferences:
[preferences.txt](https://github.com/jvpernis/esp32-ps3/files/3473930/preferences.txt)

jvpernis, 2019-08-07:
> CONFIG_BTDM_CONTROLLER_MODE_BTDM=y

Strange... This is exactly what I'm testing for in [ps3_int.h:26](https://github.com/jvpernis/esp32-ps3/blob/922fa41b8e79c9898c7ec8c7c32b1f14546a5e7e/src/include/ps3_int.h#L24).

One thing you could try is to reinstall the ESP32 boards:

- Open the Boards Manager with `Tools -> Board: "xxx" -> Boards Manager`
- Look for `esp32` (probably the last one in the list)
- Click `Remove`.
- Close the Arduino IDE (not sure if this matters)
- Clear the boards cache by deleting all files in `...AppData\Local\arduino15\staging\packages` that has anything to do with the ESP32.
- Launch the Arduino IDE
- Open the Boards Manager with `Tools -> Board: "xxx" -> Boards Manager`
- Look for `esp32` again
- Select the version you'd like to install (`1.0.1`)
- Click `Install`

Now try to compile the example sketch again. If this fixes your problem then Espressif has released updates to the same `1.0.1` version (which they do regularly with the ESP-IDF, so it wouldn't surprise me).

Sniperblue, 2019-08-07:
Unfortunately seems to be reinstalling ESP32 did not solve the issue. Still sthe same error!
My sdkconfig file and all the folder inside the sdk folder has a build date from 11.01.2019 1:35 o'clock.
Does this match with your sdk folder?

jvpernis, 2019-08-07:
> My sdkconfig file and all the folder inside the sdk folder has a build date from 11.01.2019 1:35 o'clock

Yes, this is exactly what I have as well! I still can't really wrap my head around this issue yet, but maybe someone else can chime in with their experiences.

> ... when I uncomment your check sequence ... the sketch works

Maybe I will just disable the checks when it's used as an Arduino library, as it would not really be needed anyway.

> By the way could you also change the MAC Adress in the example to `Ps3.begin("00:12:34:56:78:90")` because this was the deliving adress of my PS3 controller

I don't quite understand what you mean with this. Did you have to put `00:12:34:56:78:90` in the sketch while you wrote something else to the PS3 controller with the SixaxisPairTool?

Sniperblue, 2019-08-10:
Concerning `00:12:34:56:78:90` I did a misstake in understanding, for BT both MAC in the controllers must match. So you had to change the MAC in the ESP32 as you described in the describition if you dont want to change the MAC in the PS3 controller.

Meanwhile I playing a litte bit around with the lib and had a few question:
How can I disable one LED, set works but how can I switch them OFF?
I see the PS3controller provides  accelerometer  and gyroscope values, what is the ccommand to read out them?
What is the command to activate the rumble functionality?

I pimped a little bit your data example attached you'll find my test code:
[ESP32_BTclassic_PS3ControllerTest_v1_TXT.txt](https://github.com/jvpernis/esp32-ps3/files/3488685/ESP32_BTclassic_PS3ControllerTest_v1_TXT.txt)

jvpernis, 2019-08-10:
> How can I disable one LED, set works but how can I switch them OFF?

You can simply set the LED to 0. I might add an enum or something to make this more obvious. Your code would look like this:
```c
Ps3.setLed(0);
```

> I see the PS3controller provides accelerometer and gyroscope values, what is the ccommand to read out them?

The accelerometer and gyroscope values should be provided automatically, but I haven't tested this yet. You can read the values like this:
```c
Ps3.data.sensor.accelerometer.x;
Ps3.data.sensor.accelerometer.y;
Ps3.data.sensor.accelerometer.z;
Ps3.data.sensor.gyroscope.z;
```

> What is the command to activate the rumble functionality?

Unfortunately, this hasn't been implemented in the Arduino library yet. In the mean time, you could just use the following snippet:

```c
ps3_cmd_t cmd;

cmd.rumble_left_intensity = 0xff;
cmd.rumble_right_intensity = 0xff;
  
cmd.rumble_right_duration = 100;
cmd.rumble_left_duration = 100;
  
ps3Cmd(cmd);
```

> I pimped a little bit your data example

Thanks! I had planned to include a bigger example sketch to showcase all of the data you can access, and this helps a lot!

Sniperblue, 2019-08-10:
I include your information into the test sketch. Here are the test results:

Rumble functionality
- Only the right rumble works, the left ont won't work but I'am not sure whether this is an code or issue with my PS3 controller?

Battery functionality
- The response shows still LOW althought the Controller was loaded over night?

LED functionality
- ALL 4 LED starts blinken when the rumble request was activated and does not stop doing this?
- `Ps3.setLed(0);` means deactivation of ALL LED's so I hat to set the LED which should still active after this command again?
```
LED 1 == ON | LED 2 == OFF
Ps3.setLed(1);

LED 1 == ON | LED 2 == OFF
Ps3.setLed(0);
Ps3.setLed(2);
```

Accelerometer functionality
- Shows no values when PS3controller was moved?
Accelerometer X-Acc: -4   Y-Acc: 0   Z-Acc: 16

Gyroscope functionality
- Shows no yaw values when I rotate the PS3controller?
Gyroscope: 0

But maybe I do some misstakes during conding and the above mentioned weaknesses are no Lib issues!

jvpernis, 2019-08-10:
> But maybe I do some misstakes during conding and the above mentioned weaknesses are no Lib issues!

Oh on the contrary, I'm quite sure my library is to blame here :)

I actually haven't ever tested the accelerometer and gyroscope data, I just included what should be their data, but apparently I'm wrong with that. 

> `Ps3.setLed(0);` means deactivation of ALL LED's so I hat to set the LED which should still active after this command again?

Yes, actually, the Arduino library doesn't support having multiple LEDs turned on at once right now. To change the LED, it is therefore not required to first reset all the LEDs, you can immediately switch to LED 2:

 ```c
// LED 1 == ON | LED 2 == OFF
Ps3.setLed(1);

// LED 1 == OFF | LED 2 == ON
Ps3.setLed(2);
```

> ALL 4 LED starts blinken when the rumble request was activated and does not stop doing this?

The data to control the LEDs and to control the rumble is actually in the same packet sent to the PS3 controller. So when you sent the command to rumble, the LEDs were turned off. When I add the function `Ps3.setRumble()` to the Arduino library to control the rumble, it will take the last set LED into account.

You can do this too right now:

```c
ps3_cmd_t cmd;

cmd.led1 = true;

cmd.rumble_left_intensity = 0xff;
cmd.rumble_right_intensity = 0xff;
  
cmd.rumble_right_duration = 100;
cmd.rumble_left_duration = 100;
  
ps3Cmd(cmd);
```

> The response shows still LOW althought the Controller was loaded over night?

I actually noticed this too today, this is a bug and needs to be fixed.

> Only the right rumble works, the left ont won't work but I'am not sure whether this is an code or issue with my PS3 controller?

My PS3 controller rumbles on both sides when I use the snippet I provided, so maybe your controller is broken.

---

To summarize this thread so far, I still need to fix the following things on the Arduino library before I can release it:

- [ ] Disable the bluetooth configuration checks, as it apparently causes issues on some setups
- [ ] Add a full example sketch which showcases how to access all the data that is available from the PS3 controller
- [ ] Automatically assign LED1 to the PS3 controller upon successfull connection
- [ ] Change the input of `Ps3.setLed()` to an enum. You should be able to set multiple LEDs at once, e.g. `Ps3.setLed(Ps3::LED1 || Ps3::LED2)`
- [ ] Create a function `Ps3.resetLed()` to disable the LEDs in a similar fashion in which `Ps3.setLed()` enables them
- [ ] Create a function `Ps3.setRumble()` to enable the rumble on the PS3 controller withtout resetting the LEDs.
- [ ] Verify whether the Accelerometer and Gyroscope data works correctly
- [ ] Debug the battery level indicator, as it seems to be always reporting "LOW"

Do I miss anything? I'll probably have some spare time next weekend to work off this list!

blooop, 2019-08-11:
Hi, thanks for working on the Arduino port, its very exciting!  I'm looking forward to much better development ergonomics and compatibility with other libraries. 

Unfortunately, I was not able to get the Ps3Connect example to work, which is strange as I'm able to use your esp-idf version.   I followed all the steps including uninstalling and reinstalling the esp board for arduino.   I am able to run other esp32 examples such as the webserver so I believe the esp32 was set up correctly.

> * Select the version you'd like to install (`1.0.1`)

There is a new version 1.0.2 available now, but I installed 1.0.1 to keep the steps the same as @Sniperblue.  My configuration files match exactly.  I also tried @Sniperblue's fix to comment out some of the #ifdefs but that didn't fix it for me.

jvpernis, 2019-08-11:
> Unfortunately, I was not able to get the Ps3Connect example to work

Could you describe what is going wrong? Do you get the same compilation errors? Or does the PS3 controller simply refuse to connect?

> which is strange as I'm able to use your esp-idf version

Would you be willing to test the ESP-IDF version on the development branch of this repository? Maybe I messed something up in the meantime...

> There is a new version 1.0.2 available now, but I installed 1.0.1 to keep the steps the same as @Sniperblue.

The reason to select `1.0.1` in the steps I described was actually just to keep the configuration of @Sniperblue constant throughout the testing. In my setup I usually test with `1.0.2`. Either version should work (except for `1.0.0`), at least for me it does.

blooop, 2019-08-11:
> Could you describe what is going wrong? Do you get the same compilation errors? Or does the PS3 controller simply refuse to connect?

The code compiles and the "Ready." message prints but it never connect after that.  I have updated the arduino esp version to 1.0.2 but that doesn't fix it. 

> Would you be willing to test the ESP-IDF version on the development branch of this repository? Maybe I messed something up in the meantime...

For the esp-idf version everything works against the latest master branch commit: https://github.com/jvpernis/esp32-ps3/commit/06ab5e779c6c3cedcef048ee3203f2f868c899f9

From one of your posts to sniperblue:
> Have you changed any configuration settings using `make menuconfig`?

my menuconfig is set up to work with your esp-idf version.  I assume that would not cause problems. 

Thanks

Sniperblue, 2019-08-11:
@jvperni Concerning your working list, which had not to be finished within next week ;-)
Meanwhile I get the right rumble hardware spinning, but recognize some same things that both rumble are spinning when I send the activation command for the right one. Maybe my PS3controller is buggy but if possible you can take a look at that when you implement the rumble functionality in the Lib with your controller.
Over all I think your list include all up to now find issue! Great work...

blooop, 2019-08-11:
> > Would you be willing to test the ESP-IDF version on the development branch of this repository? Maybe I messed something up in the meantime...
> 
> For the esp-idf version everything works against the latest master branch commit: [06ab5e7](https://github.com/jvpernis/esp32-ps3/commit/06ab5e779c6c3cedcef048ee3203f2f868c899f9)

the esp-idf version also works against the latest devel commit.

jvpernis, 2019-08-12:
> For the esp-idf version everything works against the latest master branch commit: [06ab5e7](https://github.com/jvpernis/esp32-ps3/commit/06ab5e779c6c3cedcef048ee3203f2f868c899f9)

> the esp-idf version also works against the latest devel commit.

Thanks for checking! That rules a few things out at least.

> The code compiles and the "Ready." message prints but it never connect after that.

Just to sanity check:
- The MAC on your PS3 controller (I believe) is `b4:e6:2d:ee:7b:fd`
- This snippet for the esp-idf version works:
```c
uint8_t new_mac[8] = {0xb4,0xe6,0x2d,0xee,0x7b,0xfd};
ps3SetBluetoothMacAddress(new_mac);
ps3Init();
```
- This snippet for the Arduino version _doesn't_ work:
```c
Ps3.begin("b4:e6:2d:ee:7b:fd");
```

Could you try to run the example sketch with the debug level set to 'Debug'? `Tools -> Core Debug Level -> Debug`

Maybe this will give some insight as to why it fails to connect.

navar00, 2019-08-12:
it works great to me! first time right! hurra!

blooop, 2019-08-12:
I've fixed it.    In the esp-idf version I did not set the bluetooth address using ```ps3SetBluetoothMacAddress``` because I just set the ps3 controller to the native mac address of the esp32.   My init code was simply:

```
 ps3SetEventCallback(controller_event_cb);
 nvs_flash_init();
 ps3Init();
```

In the arduino version I had:

```
Ps3.begin("b4:e6:2d:ee:7b:fd");
```

The fix was to use the sixaxis tool to set the address to "00:12:34:56:78:90" and update the ```Ps3.begin( "00:12:34:56:78:90" )``` to match.

> Could you try to run the example sketch with the debug level set to 'Debug'? `Tools -> Core Debug Level -> Debug`

This probably doesn't matter any more but I was not able to get debug mode working even after changing the debug level.  I had a look online to try and get debug info and added:
```
Serial.setDebugOutput(true);
esp_log_level_set("*", ESP_LOG_DEBUG);
```
This also didn't work, but the problem is fixed now so everything is fine. 

Thanks for your help!

blooop, 2019-08-12:
False alarm, when I added some bad code code I get debug messages. E.g.:
```
E (6767) PS3_GAP: [ps3_gap_send_hid] sending command: failed
```

I guess the non connection state was not enough to log a debug message.

jvpernis, 2019-08-24:
> Accelerometer functionality
> 
>     * Shows no values when PS3controller was moved?

I've crossed a few items off my list, such as investigating the accelerometer values. They appear to be working just fine with me. 

To debug this, I created an example sketch to display the accelerometer data. Could you try it out? You can simply download the  [esp32-ps3-develop.zip](https://codeload.github.com/jvpernis/esp32-ps3/zip/develop) again and replace the old one.

This is what I get when I run the sketch:

[image]

Sniperblue, 2019-08-26:
My controller model CECHZC2U A1 still shows -4 0 16 by using your updatet lib. Maybe controller issue...

PunyViking, 2019-11-10:
Thanks for making this library for Arduino too! I love it, I only wished it would work for ps4 controllers as all my ps3 controllers are broken......  

I can connect my esp32 to the ps4 controller, but I get strange readings from the digital buttons and the analog sticks.

But thanks for your work so far, its the best alternative I've found so far!!

jvpernis, 2019-11-10:
Thanks for the kind words! There is actually a PS4 controller owned by a friend of mine on my desk waiting for me to implement it in this library! The next big challenge after finding some spare time to work on this would be to find a name that better suits this library, though :)

PunyViking, 2019-11-10:
Awesome! Please let me know if you need help betatesting it! I'm not good with software, but I will help with what I can.

I tried to connect the ps4 controller to my d2-mini-esp32 with this library, and it kinda works. Its just the parsing/button mapping that is totally wrong to make it work with basic functions. 

The library should be called "The best and only library you will need in the world". There is a lot of people looking for a way to connect the ps4 controller to the esp32, you will be a hero for many! :-D

klimbot, 2019-11-14:
Not sure if this is direct result of the library, but when I'm running Ps3.begin() I'm getting the error:
```
E (263747) PS3_GAP: ps3_gap_init_service Registering GAP service PS3-HIDC failed
E (263747) PS3_GAP: ps3_gap_init_service Registering GAP service PS3-HIDI failed
```
The function works as expected when running in a small program but when running in a more complex program it's throwing these errors.
Are any special timers/registers or something required that might already be in use?

YannP84, 2020-01-12:
Hello, Thanks for your work,  I want to remote my ballancing robot with a PS3 controller. 
I'm testing "Ps3Connect exemple" In arduino IDE, My controller is well connected but there is no comunication with the datas. axis always at zero, and no button press notification
The monitor tell me : "the controller baterry charge is undefined" and the controller is always bliking.

Here are the logs 
[logs.txt](https://github.com/jvpernis/esp32-ps3/files/4051392/logs.txt)

Did I have to use ESP-IDf to modify the bluetooth protocol BLE=>BR/EDR ?
The mac adress of BT ESP32 is the same that is paired in Sisxaxispairtool.

Thank you very much

The controller is working fine with my PS3.

Have you any idea ?

bennellis, 2020-01-17:
Hello, I have gotten your library working on my esp, and the four buttons work perfectly (cross, triangle, circle and square). I am just wondering about the functions for the analog sticks on the ps3 remote and the other buttons as well. I am not sure where I would find these functions as I am quite new to Arduino IDE. Any help would be greatly appreciated.

jvpernis, 2020-01-25:
It has been done! I have requested this library to be included in the Arduino Library Manager. 

You can follow this request in Arduino issue [#9663](https://github.com/arduino/Arduino/issues/9663).

I will close this issue once the library is available from within the Arduino IDE.

closed 2020-03-10 by jvpernis

jlmyra, 2022-05-26:
The code looks like this:
ps3_cmd_t cmd;

    cmd.rumble_left_intensity = 0x7d;
    cmd.rumble_right_intensity = 0x7d; //0xff = 255
  
    cmd.rumble_right_duration = 100;
    cmd.rumble_left_duration = 100;
  
    ps3Cmd(cmd);

Sriram510, 2022-05-26:
> The code looks like this: ps3_cmd_t cmd;
> 
> ```
> cmd.rumble_left_intensity = 0x7d;
> cmd.rumble_right_intensity = 0x7d; //0xff = 255
> 
> cmd.rumble_right_duration = 100;
> cmd.rumble_left_duration = 100;
> 
> ps3Cmd(cmd);
> ```

when to use ps3_cmd_t cmd; and ps3Cmd(cmd); in the code?

Sriram510, 2022-05-26:
I wonder if we can control the 4 lights on the controller ,if yes how?

jlmyra, 2022-05-26:
See line 213 of Ps3Demo in the Examples. 
4 lights - setPlayer(10)
3 lights - setPlayer(9)
2 lights - setPlayer(7)

There is an Example sketch for the rumble feature in the Examples.

Sriram510, 2022-05-31:
Respected sir,

I am unable to connect my ps3 controller to esp32.What's happening is all 5 leds blink simultaneously and still never get connected. If you could explain how the connection between the two is made and how to read the ps3 data. As I am an amateur if you the could provide the necessary lines of code to establish connection and read data and able to rumble and change led lights, would be of great help. I request you sir please help me with this.

Thank you,
 Regards,
sriram