jvpernis/esp32-ps3 #4: I wished it would work for ps4 controllers as all my ps3 controllers are broken......

jvpernis, 2019-11-15:
Thanks for making this library for Arduino too! I love it, I only wished it would work for ps4 controllers as all my ps3 controllers are broken......  

I can connect my esp32 to the ps4 controller, but I get strange readings from the digital buttons and the analog sticks.

But thanks for your work so far, its the best alternative I've found so far!!

_Originally posted by @PunyViking in https://github.com/jvpernis/esp32-ps3/issues/3#issuecomment-552197754_

PunyViking, 2019-12-30:
Somebody just did it! 

https://www.psxhax.com/threads/ps4-esp32-use-a-playstation-4-controller-with-esp32-by-nurobotics.7246/?fbclid=IwAR10AC9hno936FAu9-VH_8HAPaqgHr89yHWEnGsChKHY9LeHS8BiztrZHWU

grasmanek94, 2020-05-22:
I think the new link is  https://github.com/openobjects/PS4-esp32

DerfJagged, 2020-06-23:
FYI, ESP-IDF now includes a bluetooth HID host example for Classic bluetooth which works with PS4 out of the box. It can cold connect without having to supply a MAC address, and it also reconnects!

https://github.com/espressif/esp-idf/tree/release/v4.2/examples/bluetooth/esp_hid_host

@jvpernis just letting you know in case it helps make your development easier. No idea if it supports multiple controllers at once.

iotPeanut, 2020-08-07:
> FYI, ESP-IDF now includes a bluetooth HID host example for Classic bluetooth which works with PS4 out of the box. It can cold connect without having to supply a MAC address, and it also reconnects! [...]

Yes, I have also tested it. It can be connected and can communicate normally. But do you know how to port this program to ardunio. I try to compile esp_idf/components /bt and esp_idf /components/esp_hid into library functions in IDF programming mode, and then put them under the SDK folder of Arduino. But failed, do you have a good idea, thank you

DerfJagged, 2020-08-07:
@iotPeanut I don't unfortunately. Please let me know if you do figure it out!

freevedo, 2023-04-19:
I had succeed the connection to establish but I can't read any data from the PS4 controller. How does it work ? I'm new with C++