jvpernis/esp32-ps3 #5: When I'm running Ps3.begin() I'm getting the error: Registering GAP service [...] failed

jvpernis, 2019-11-15:
Not sure if this is direct result of the library, but when I'm running Ps3.begin() I'm getting the error:
```
E (263747) PS3_GAP: ps3_gap_init_service Registering GAP service PS3-HIDC failed
E (263747) PS3_GAP: ps3_gap_init_service Registering GAP service PS3-HIDI failed
```
The function works as expected when running in a small program but when running in a more complex program it's throwing these errors.
Are any special timers/registers or something required that might already be in use?

_Originally posted by @klimbot in https://github.com/jvpernis/esp32-ps3/issues/3#issuecomment-553830506_

jvpernis, 2019-11-15:
In the more complex program, what components do you use? Is there perhaps another part of your code that uses the bluetooth functionality?

klimbot, 2019-11-15:
No, it's a project ported from a more conventional Arduino board like the atmega that I'm trying to get running with the ESP32. It messes with registers, interrupts and timers so I was wondering if there were any particular requirements of this library

klimbot, 2019-11-22:
Found the issue: I was calling Ps3.begin too many times by accident. 

Issue can be closed, thanks

jvpernis, 2020-01-25:
Thanks for letting me know! If you need help with anything else, feel free to open another issue.

closed 2020-01-25 by jvpernis