jvpernis/esp32-ps3 #10: Chinese PS3 controller

beckmx, 2020-01-23:
Hello, just a doubt, I would like to see the raw data coming to the bluetooth, I have a clone controller but none of the configurations done is working with this controller thus, I would like to know if there is any way to see the raw data

Testato, 2020-04-01:
I also have a chinease clone, on plsystation consolle it work, but i have problem when try to use it on PC or on Smartphone.
Maybe it do not mantain in memory the mac address ?

@jvpernis 
Thanks a lot for your library it is a brillant idea, i will test it with an official and a chinease ps3 pad

bivekSapkota, 2020-07-17:
Hi, I just bought some clone controllers for my robotics project. But the code does not seem to work with the controller. The controller works fine with sixaxis pair tool in both PC and Android although it shows as fake controller. I have set the mac address in both the code and controller to "00:01:02:03:04:05" but only thing the controller does is blink all four leds indefinitely as long as the esp32 (with your code) is powered. Blinking stops after the esp32 is powered off.  The Serial monitor shows "Ready." but after that nothing happens. I hope you will help solve the issue soon. Thanks.

welverin, 2021-01-20:
same here, controller connected to pc windows, Mac and ps4 but not to esp32

welverin, 2021-01-27:
i test a new one and it works, probably some clone is more clone than other :)

tarasradio, 2021-01-29:
I have the same problem with my chinese PS3 Controller (When I try to pair this one to ESP, It is blinking for all LEDs and then waits some seconds to off).

I have tried printed the debug info from libraries files, and I find that the joystick tries to connect to ESP and connection stopped on the first call ps3_l2cap_data_ind_cback (ps3_l2cap.c). In this method I printed the pbuf->len and it equal 1 but it should be equal or more than 2... and it's means that this call doesn't calls the parser and parser doesn't call the connection callback and e.t.c.

tarasradio, 2021-01-29:
[image]
It is my debug log with my own includes (I have tried to call the parser if the p_buf->len <=2 but it hadn't be usefull). 
Also, in my log I printed 'event' field from p_buf and its equal 4352 (dec) | 0x1100 (hex). I would like to have a help to understand it's means...

Naruto128, 2021-02-05:
> Hi, I just bought some clone controllers for my robotics project. But the code does not seem to work with the controller. The controller works fine with sixaxis pair tool in both PC and Android [...]

Hello, I'm the same. my jostick was sold to me as original, the truth is difficult to know if it is original or not. any solution for this please

tarasradio, 2021-02-05:
> Hello, I'm the same. my jostick was sold to me as original, the truth is difficult to know if it is original or not. any solution for this please

I work on it... But I feel that it will be too hard =) I going to find original controller and compare it's packets with chinese controller's packets

Testato, 2021-02-06:
But all yours fake controller wok on original.sony consolle ?
Mine work, i think.this is an important info to remember, probably the PS bypass some check ?

For example mine fake controller not do not store the macaddress, it is always 0.0.0.0 but it work.on the play station

Naruto128, 2021-02-07:
> Hello, I'm the same. my jostick was sold to me as original, the truth is difficult to know if it is original or not. any solution for this please

 have found this library but it is for ps4 jostick, but it is very similar to the ps3 one. I miss it that I tried it with my ps3 control, and it shows me that it has connected, but it does not throw me any of the control status.

https://github.com/aed3/PS4-esp32

chodzi, 2021-03-09:
hi, I'm using the ESP-IDF library with sixaxis CECHZC2U-a1 Chinese PS3 controller. The communication with the controller started ok and I can identify the buttons, but the controller is not synchronized, all 4 leds are blinking and after few seconds it's turns off. I will appreciate any useful help.

```
I (5493591) PS3_L2CAP: [ps3_l2cap_connect_ind_cback] bd_addr: �Z\M
  l2cap_cid: 0x40
  psm: 17
  id: 1
I (5493601) PS3_L2CAP: [ps3_l2cap_config_ind_cback] l2cap_cid: 0x40
  p_cfg->result: 0
  p_cfg->mtu_present: 0
  p_cfg->mtu: 0
I (5493601) PS3_L2CAP: [ps3_l2cap_config_cfm_cback] l2cap_cid: 0x40
  p_cfg->result: 0
I (5493621) PS3_L2CAP: [ps3_l2cap_connect_ind_cback] bd_addr: �Z\M
  l2cap_cid: 0x41
  psm: 19
  id: 3
I (5493631) PS3_L2CAP: [ps3_l2cap_config_ind_cback] l2cap_cid: 0x41
  p_cfg->result: 0
  p_cfg->mtu_present: 0
  p_cfg->mtu: 0
I (5493641) PS3_L2CAP: [ps3_l2cap_config_cfm_cback] l2cap_cid: 0x41
  p_cfg->result: 0
I (5493641) PS3_L2CAP: [ps3_l2cap_send_hid] sending command: success
The user released the l1 button
The user started pressing the triangle button
The user released the triangle button
I (5503711) PS3_L2CAP: [ps3_l2cap_disconnect_ind_cback] l2cap_cid: 0x40
  ack_needed: 0
I (5503711) PS3_L2CAP: [ps3_l2cap_disconnect_ind_cback] l2cap_cid: 0x41
  ack_needed: 0
```