jvpernis/esp32-ps3 #11: Hello, Thanks for your work,  I want to remote my ballancing robot with a PS3 controller.

jvpernis, 2020-01-25:
Hello, Thanks for your work,  I want to remote my ballancing robot with a PS3 controller. 
I'm testing "Ps3Connect exemple" In arduino IDE, My controller is well connected but there is no comunication with the datas. axis always at zero, and no button press notification
The monitor tell me : "the controller baterry charge is undefined" and the controller is always bliking.

Here are the logs 
[logs.txt](https://github.com/jvpernis/esp32-ps3/files/4051392/logs.txt)

Did I have to use ESP-IDf to modify the bluetooth protocol BLE=>BR/EDR ?
The mac adress of BT ESP32 is the same that is paired in Sisxaxispairtool.

Thank you very much

The controller is working fine with my PS3.

Have you any idea ?

_Originally posted by @YannP84 in https://github.com/jvpernis/esp32-ps3/issues/3#issuecomment-573454628_

jvpernis, 2020-03-10:
Hi @YannP84 are you still having difficulties with the library?

As of February this year the library has been available to install from within the Arduino IDE. Could you please try to install this library though the Arduino IDE and run the Ps3Data sketch? It is called `PS3 Controller Host` in the Arduino Library Manager. Thanks!

Naruto128, 2021-02-03:
Hello, thank you very much for the bookstore.

i'm trying to get my jostick ps3 to work, but i can't.

It compiles well but it just doesn't connect, the lights don't stop flashing, the strange thing is that if I reset the esp32, the control in a few seconds turns off. that is, it kind of connects partially, but it never tells me that it has successfully connected.

I have tried it to jostick with my pc, with cable and saw bluetooth to play games and it works correctly

I'm using
ide arduino 1.8.10
ide esp32 1.0.1

PS3_Controller_Host, I have tried all versions.

Naruto128, 2021-02-04:
[url=https://subefotos.com/ver/?1c41f19008ca20912858e625fe2ac360o.png][img]http://thumbs.subefotos.com/1c41f19008ca20912858e625fe2ac360o.jpg[/img][/url]

Naruto128, 2021-02-04:
https://subefotos.com/ver/?1c41f19008ca20912858e625fe2ac360o.png

Naruto128, 2021-02-05:
hi??

Naruto128, 2021-02-07:
I have found this library but it is for ps4 jostick, but it is very similar to the ps3 one. I miss it that I tried it with my ps3 control, and it shows me that it has connected, but it does not throw me any of the control status.

https://github.com/aed3/PS4-esp32