jvpernis/esp32-ps3 #12: Hello, I have gotten your library working on my esp, and the four buttons work perfectly (cross, triangle, circle and square). I am just wondering about the functions for the analog sticks on the ps3 remote and the other buttons as well. I am not sure where I would find these functions as I am quite new to Arduino IDE. Any help would be greatly appreciated.

jvpernis, 2020-01-25:
Hello, I have gotten your library working on my esp, and the four buttons work perfectly (cross, triangle, circle and square). I am just wondering about the functions for the analog sticks on the ps3 remote and the other buttons as well. I am not sure where I would find these functions as I am quite new to Arduino IDE. Any help would be greatly appreciated.

_Originally posted by @bennellis in https://github.com/jvpernis/esp32-ps3/issues/3#issuecomment-575793841_

jvpernis, 2020-01-25:
Hi @bennellis, I have added a new example sketch called `Ps3Demo`, which showcases how to access most of the data provided by this library.

Could you let me know whether this example sketch is sufficient or if you'd prefer to have more examples? Thanks!

bennellis, 2020-02-27:
Hi @jvpernis, that sketch works perfectly thanks! New question, I have been trying to connect two Ps3 controllers to one Esp32, and have not had much success. Do you know of any way to do this with your library? Thanks!

jvpernis, 2020-03-10:
Thanks for letting me know! I see you've created a separate issue for your next question, so I'll be closing this one for now. Feel free to reopen this issue in case you still need help!

closed 2020-03-10 by jvpernis