jvpernis/esp32-ps3 #14: Two ps3 controllers?

bennellis, 2020-03-04:
Hi, I’m wondering if you know of any way to hook up two ps3 controllers to one esp32 using your library? Thanks!

jvpernis, 2020-03-10:
Hi, unfortunately, this library doesn't support handling multiple controllers at once at the moment. In order to support such a thing, I'd need to change quite a bit in this library and provide some kind of an interface to separate the two controllers. 

I'll keep this issue open until I have gotten around to implementing it :)

mlorenzati, 2020-07-12:
Love your project jvpernis, I'm on the same pages as bennellis, I've built a unijoysticle like board (to control a commodore with bluetooth) and right now that code does not support DS3, so I'm using your library to use one DS3, looking to have both.

fredlcore, 2021-03-08:
Hi, I would also like to thank you for this project and add me to the list of users looking forward to using two controllers with your library. I had initially thought that simply instantiating a second object might do the trick, but it seems that it's not so easy...

jlvandusen, 2021-05-17:
@jvpernis I too am looking for 2 nav controller support for my droid projects... I need both for the 2 analog controllers separated by my hands...  the nav controllers were perfect for this

doragasu, 2021-07-14:
This feature would be really nice!

digitalbird01, 2021-07-28:
Great work, I was wondering is it possible for the Ps3 controller to communicate with two ESP32 boards at the same time?
I am working on a Camera motion control project which consists of a pan/tilt head and separate camera slider each with its own esp32. I have successfully used your library to control the parts individually but I need the controller to connect with both Esp32 boards at the same time is this possible?

RaspberryPiDude314, 2023-08-07:
>I have successfully used your library to control the parts individually but I need the controller to connect with both Esp32 boards at the same time is this possible?

If anyone else is still curious about this, the simple solution is to connect via Bluetooth to one of the boards, and connect the two with serial, or an HC-12 or similar if wireless is required (maybe ESPNow if it supports bluetooth use at the same time, I don't know about that). Most bluetooth devices simply don't support connecting to 2 devices at the same time, and this will cause problems for timing or similar in any situation where both boards don't get the same data from the controller. An example is if there is a bad connection on just one controller and not the other.