jvpernis/esp32-ps3 #15: Arduino 1.8.12 Compilation Error

fbiego, 2020-03-10:
```
C:\Users\user\Documents\Arduino\libraries\esp32-ps3-master\src\Ps3Controller.cpp: In member function 'bool Ps3Controller::begin(char*)':

C:\Users\user\Documents\Arduino\libraries\esp32-ps3-master\src\Ps3Controller.cpp:53:63: error: format '%x' expects argument of type 'unsigned int*', but argument 3 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]

     if (sscanf(mac, ESP_BD_ADDR_STR, ESP_BD_ADDR_HEX_PTR(addr)) != ESP_BD_ADDR_LEN){

                                                               ^

C:\Users\user\Documents\Arduino\libraries\esp32-ps3-master\src\Ps3Controller.cpp:53:63: error: format '%x' expects argument of type 'unsigned int*', but argument 4 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
```

fbiego, 2020-03-10:
Changed all `uint8_t*` to `unsigned int*` in `Ps3Controller.cpp` line 52 but got new error

```
C:\Users\user\Documents\Arduino\libraries\esp32-ps3-master\src\Ps3Controller.cpp: In member function 'bool Ps3Controller::begin(char*)':

C:\Users\user\Documents\Arduino\libraries\esp32-ps3-master\src\Ps3Controller.cpp:53:15: error: array subscript is above array bounds [-Werror=array-bounds]

     if (sscanf(mac, ESP_BD_ADDR_STR, ESP_BD_ADDR_HEX_PTR(addr)) != ESP_BD_ADDR_LEN){

               ^

C:\Users\user\Documents\Arduino\libraries\esp32-ps3-master\src\Ps3Controller.cpp:53:15: error: array subscript is above array bounds [-Werror=array-bounds]
```

jvpernis, 2020-03-10:
Hi fbiego, thanks for using this library. Could you share a little bit more about your current setup?

- What esp32 boards library version do you use in your Arduino IDE?
- Have you installed my esp32-ps3 library through Arduino's Library Manager?

I have just tried compiling an example sketch using esp32 version 1.4 and Arduino version 1.8.12 (both the most recent versions), but I don't get the same error message, so I haven't been able to reproduce this issue...

fbiego, 2020-03-11:
Hello @jvpernis, I didn't install through the Library Manager, I will try reinstalling and give you feedback on the issue.

fbiego, 2020-03-11:
Reinstalled using Arduino's Library Manager but I still get the same errors.
Did you mean esp32 version 1.0.4? That's what I have.

The Board is **DOIT ESP32 DEVKIT V1**

I have attached the error log below.

[error.txt](https://github.com/jvpernis/esp32-ps3/files/4318955/error.txt)

closed 2020-03-18 by fbiego

kawashimaken, 2020-07-29:
[error.txt](https://github.com/jvpernis/esp32-ps3/files/4995132/error.txt)
I have the similar error, I am using Arduino IDE 1.8.13
```
/Users/demo/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3.c: In function 'ps3SetLed':
/Users/demo/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3.c:160:5: error: suggest parentheses around assignment used as truth value [-Werror=parentheses]
     if(cmd.led4 = player >= 4) player -= 4;
     ^
/Users/demo/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3.c:161:5: error: suggest parentheses around assignment used as truth value [-Werror=parentheses]
     if(cmd.led3 = player >= 3) player -= 3;
     ^
/Users/demo/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3.c:162:5: error: suggest parentheses around assignment used as truth value [-Werror=parentheses]
     if(cmd.led2 = player >= 2) player -= 2;
     ^
/Users/demo/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3.c:163:5: error: suggest parentheses around assignment used as truth value [-Werror=parentheses]
     if(cmd.led1 = player >= 1) player -= 1;
     ^
cc1: some warnings being treated as errors
```
I have tried several different  ESP32 boards, the errors are the same I guess it is not due to certain board.

kawashimaken, 2020-07-29:
By modifying in the preferencess-> compiler warnigs-> to default
makes the error go away.