jvpernis/esp32-ps3 #24: Error compiling library

djmrozinski, 2020-12-29:
Initially I was getting a else without if error when I attempted to compile.  Error was in ps3_gap.c file on line 107.  I noticed there weren't any {} with the if so I added.  That seemed to make that error go away.

Now when I compile I am getting the following errors with all the examples included with the library.
Any suggestions???

Thanks
Dave

Arduino: 1.8.14 Hourly Build 2020/12/15 11:34 (Mac OS X), Board: "ESP32 Dev Module, DIO, 80MHz, 4MB (32Mb), 921600, None"
libraries/esp32-ps3-master/ps3_gap.c.o:(.literal.ps3_gap_init_service+0x1c): undefined reference to `GAP_ConnOpen'
libraries/esp32-ps3-master/ps3_gap.c.o:(.literal.ps3_gap_event_handle+0xc): undefined reference to `GAP_ConnGetL2CAPCid'
libraries/esp32-ps3-master/ps3_gap.c.o:(.literal.ps3_gap_event_handle+0x10): undefined reference to `GAP_ConnBTRead'
libraries/esp32-ps3-master/ps3_gap.c.o:(.literal.ps3_gap_send_hid+0x14): undefined reference to `GAP_ConnBTWrite'
libraries/esp32-ps3-master/ps3_gap.c.o: In function `ps3_gap_init_service':
/Users/djmrozinski/Documents/Arduino/libraries/esp32-ps3-master/src/ps3_gap.c:60: undefined reference to `GAP_ConnOpen'
libraries/esp32-ps3-master/ps3_gap.c.o: In function `ps3_gap_update_connected':
/Users/djmrozinski/Documents/Arduino/libraries/esp32-ps3-master/src/ps3_gap.c:60: undefined reference to `GAP_ConnGetL2CAPCid'
/Users/djmrozinski/Documents/Arduino/libraries/esp32-ps3-master/src/ps3_gap.c:60: undefined reference to `GAP_ConnGetL2CAPCid'
libraries/esp32-ps3-master/ps3_gap.c.o: In function `ps3_gap_event_handle':
/Users/djmrozinski/Documents/Arduino/libraries/esp32-ps3-master/src/ps3_gap.c:60: undefined reference to `GAP_ConnBTRead'
libraries/esp32-ps3-master/ps3_gap.c.o: In function `ps3_gap_send_hid':
/Users/djmrozinski/Documents/Arduino/libraries/esp32-ps3-master/src/ps3_gap.c:103: undefined reference to `GAP_ConnBTWrite'
libraries/esp32-ps3-master/ps3_spp.c.o:(.literal.ps3_spp_callback+0x8): undefined reference to `esp_bt_gap_set_scan_mode'
libraries/esp32-ps3-master/ps3_spp.c.o:(.literal.ps3_spp_callback+0xc): undefined reference to `esp_spp_start_srv'
libraries/esp32-ps3-master/ps3_spp.c.o:(.literal.ps3_spp_init+0x14): undefined reference to `esp_spp_register_callback'
libraries/esp32-ps3-master/ps3_spp.c.o:(.literal.ps3_spp_init+0x18): undefined reference to `esp_err_to_name'
libraries/esp32-ps3-master/ps3_spp.c.o:(.literal.ps3_spp_init+0x1c): undefined reference to `esp_spp_init'
libraries/esp32-ps3-master/ps3_spp.c.o: In function `ps3_spp_callback':
/Users/djmrozinski/Documents/Arduino/libraries/esp32-ps3-master/src/ps3_spp.c:103: undefined reference to `esp_bt_gap_set_scan_mode'
/Users/djmrozinski/Documents/Arduino/libraries/esp32-ps3-master/src/ps3_spp.c:106: undefined reference to `esp_spp_start_srv'
libraries/esp32-ps3-master/ps3_spp.c.o: In function `ps3_spp_init':
/Users/djmrozinski/Documents/Arduino/libraries/esp32-ps3-master/src/ps3_spp.c:68: undefined reference to `esp_spp_register_callback'
/Users/djmrozinski/Documents/Arduino/libraries/esp32-ps3-master/src/ps3_spp.c:69: undefined reference to `esp_err_to_name'
/Users/djmrozinski/Documents/Arduino/libraries/esp32-ps3-master/src/ps3_spp.c:73: undefined reference to `esp_spp_init'
/Users/djmrozinski/Documents/Arduino/libraries/esp32-ps3-master/src/ps3_spp.c:74: undefined reference to `esp_err_to_name'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

jlmyra, 2020-12-30:
I experienced a similar problem. I am using Arduino IDE 1.8.13 with MH-ET ESP32 Mini. If I set the _Arduino>Preferences>Compiler Warning>Default and Tools>Core Debug Level>None_ my sketch will compile. Other combinations (_Arduino>Preferences>Compiler Warning>More or All_) will throw errors and compilation will fail.

djmrozinski, 2020-12-31:
I was able to resolve the problem.  I had recently upgraded to 1.8.14 and was trying to compile the sketch with the new upgrade.  What  I did was reinstall all my board definitions and libraries.  Once that was done, the program compiled.

Thanks
Dave

fabo311, 2021-01-09:
> I was able to resolve the problem. I had recently upgraded to 1.8.14 and was trying to compile the sketch with the new upgrade. What I did was reinstall all my board definitions and libraries. Once that was done, the program compiled. Thanks Dave
> […](#)

I have the same issue, can you help me how to reinstall all my board definitions and libraries?
Thanks,
Fabo

jvpernis, 2021-01-26:
Hi!

The errors you were getting are fixed in the new version of this library, could you please try again to see if this has resolved your issue?

Thanks!