jvpernis/esp32-ps3 #26: ERROR COMPILING

moiss21, 2021-01-16:
C:\Users\Moisés\Desktop\PROYECTOS MAKER\PS3 + SERVO\Ps3Demo\Ps3Demo.ino: In function 'void setup()':
C:\Users\Moisés\Desktop\PROYECTOS MAKER\PS3 + SERVO\Ps3Demo\Ps3Demo.ino:203:34: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
C:\Users\Mois�s\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3.c: In function 'ps3SetLed':
C:\Users\Mois�s\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3.c:160:5: error: suggest parentheses around assignment used as truth value [-Werror=parentheses]
     if(cmd.led4 = player >= 4) player -= 4;
     ^
C:\Users\Mois�s\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3.c:161:5: error: suggest parentheses around assignment used as truth value [-Werror=parentheses]
     if(cmd.led3 = player >= 3) player -= 3;
     ^
C:\Users\Mois�s\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3.c:162:5: error: suggest parentheses around assignment used as truth value [-Werror=parentheses]
     if(cmd.led2 = player >= 2) player -= 2;
     ^
C:\Users\Mois�s\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3.c:163:5: error: suggest parentheses around assignment used as truth value [-Werror=parentheses]
     if(cmd.led1 = player >= 1) player -= 1;
     ^
cc1.exe: some warnings being treated as errors
exit status 1
Error compilando para la tarjeta ESP32 Dev Module.

moiss21, 2021-01-16:
I need a solution asap, please.

jlmyra, 2021-01-16:
I experienced a similar problem. I am using Arduino IDE 1.8.13 with MH-ET ESP32 Mini. If I set the Arduino>Preferences>Compiler Warning>Default and Tools>Core Debug Level>None my sketch will compile. Other combinations (Arduino>Preferences>Compiler Warning>More or All) will throw errors and compilation will fail.

moiss21, 2021-01-16:
The error continues. I read your settings before, i try everything about how to resolve this error but i cant.
Maybe another solution?

moiss21, 2021-01-16:
When i compile for arduino the code says to me he cant found, for example, the bt main.h

moiss21, 2021-01-16:
In diferent boards the errro is diferent, but i think this maybe only one error, that affects to all boards, obviously. 
I dont know if i need a new library, or there some error on the code, or i dont know.

moiss21, 2021-01-16:
[image]

jlmyra, 2021-01-16:
If it can't find bt_main.h then it is not in your libraries folder. Try installing the PS3 library from Arduino>Sketch>Include Library and search for PS3 Controller Host.

moiss21, 2021-01-16:
It says undefinied refrence to GAP etc, do you read it? I dont know what is it.
And, i cant select Sketch options, its empty.
[image]

moiss21, 2021-01-16:
If i select all warnings in PREFERENCES -> COMPILATOR, it shows me another errros, 
[image]

moiss21, 2021-01-16:
I dont wanna compile for arduino,but if i do it it shows that error, about esp.bt.main.h, which is in the sdk - include - bt
[image]
 folder

jlmyra, 2021-01-16:
I'm not sure what else I can add. I know that if you have error reporting turned on(in preferences) the compiler will assume that warnings are errors and stop the compilation. Turn the warnings off. You might consider uninstalling and reinstalling the Arduino IDE and your libraries.

moiss21, 2021-01-16:
Yes, i do it like 3 times today, the preferences in none warnings ant reistaling all libraries and espressifs datas, and unistallinf ide.... i dont know why it dont compile.

jvpernis, 2021-01-16:
I haven't worked with this library for a long while, but I have just tried to compile my example sketch `Ps3Connect` to see if it still works, with these software versions:
|Component|Version|
|-------------|--------|
|Arduino IDE|1.8.13|
|esp32 boards|1.0.4|
|PS3 Controller Host|1.0.0|

With the compiler warnings set to `All`, I'm getting the same errors as you:

[image]

However, when I set the warnings to `None`, the sketch compiles just fine:

[image]

You can disable the warnings by going to:

- On the toolar, select `File` (in your case `Archivo`)
- Then `Preferences` (in your case `Preferencias`)
- In the dropdown `Compiler warnings`
- Select `None`

[image]

Please share which versions of the software you are using, and what settings you're using. I don't see any errors coming from the libraries (only warnings which probably should be fixed in a next version).

moiss21, 2021-01-16:
Arduino IDE 1.8.13
Ps3 Controller 1.0.0
And i use the ESP espressif data, because my esp32 is this:
[image]

moiss21, 2021-01-16:
And this is my preferences. I just delete all many times today, and yesterday, and a simple code like his dont compile. 

It throw me errors like  - undefined reference to `GAP_ConnOpen' -
I dont know why

[image]

jvpernis, 2021-01-16:
I have noticed that you don't have any Board Manager URLs configured. Did you download the ESP32 boards from their GitHub?

If so, could you try to add their board manager URL to the Arduino IDE instead?:

- https://dl.espressif.com/dl/package_esp32_index.json

And install the `esp32` version 1.0.4:

[image]

moiss21, 2021-01-16:
I knew you were going to answer that to me, so that's what I'm doing, a few hours ago I downloaded the esp8266 board preferences, which obviously didn't work for me, I didn't know there were ones from esp32.

I have to say that I have always used espressif settings.
Let's see if using these versions that you tell me works for me.

jvpernis, 2021-01-16:
The reason I'm asking you is because I'm quite aware that the library is not going to be compatible with their upcoming release 1.0.5. When you download the boards from the GitHub directly, you are getting their latest software, which currently is 1.0.5-rc. This version is not available in the Arduino IDE.

moiss21, 2021-01-16:
Oh man, thank you so much.
I've been looking for information for literally two whole days, modifying your .h and .cpp libraries ... reinstalling everything, deleting everything ...

It already compiles and works, I have to thank you for your work, both for the library and for helping me with my failure.

Thank you !
Greetings from Spain.

moiss21, 2021-01-16:
One doubt, how i can put off the bluettoth controller?
There any contrary code to set off the controller?

Ps3.isConnected()
Ps3.attachOnConnect(onConnect);

Like Ps3.isnotConnected or Ps3.attachOffConnect

A line code to expulse the conexion without reseting the esp.

jvpernis, 2021-01-16:
Glad to know it now compiles properly now!

Unfortunately, as you can see in [Ps3Controller.cpp:67](https://github.com/jvpernis/esp32-ps3/blob/master/src/Ps3Controller.cpp#L67), that's something I still need to implement. I took a quick look at Espressif's [SimpleBLE example sketch](https://github.com/espressif/arduino-esp32/blob/master/libraries/SimpleBLE/src/SimpleBLE.cpp#L94), and it looks like you need something like this to disable the bluetooth:

```c
if(btStarted()){
    esp_bluedroid_disable();
    esp_bluedroid_deinit();
    btStop();
}
```

moiss21, 2021-01-17:
I guess to turn it back on it would be something like

```
if (! btStarted ()) {
    esp_bluedroid_enable ();
    esp_bluedroid_init ();
    btStart ();
}
```

I must thank you for your help, now I can continue my project in a simpler way, and for solving this last question too, thank you very much.