jvpernis/esp32-ps3 #28: ESP32 arduino software version

hreintke, 2021-01-25:
Hi,

Did some tests and successfully used your library a couple of months ago.
As I am now going to use it in my project I was checking current status and noticed this remark. 

> The reason I'm asking you is because I'm quite aware that the library is not going to be compatible with their upcoming release 1.0.5. When you download the boards from the GitHub directly, you are getting their latest software, which currently is 1.0.5-rc6. This version is not available in the Arduino IDE.

Does that mean that this library cannot be used at all with newer versions (1.05 and upcoming IDF4) or is it temporary and will those be supported by upcoming updates ?

jvpernis, 2021-01-26:
Hi,

This library has to use Espressif's internal APIs to handle the Bluetooth connectivity, as the "public" APIs do not expose enough control for use with a PS3 controller. The SDK provided by Espressif (ESP-IDF) provides you with tools to compile these internal APIs or not through the `sdkconfig` file (as explained in [this section of the README](https://github.com/jvpernis/esp32-ps3#configuring-your-project)). 

However, for the ESP32 board definitions in the Arduino IDE, Espressif uses precompiled binaries of the ESP-IDF SDK to prevent long compile times for Arduino IDE users. This was no issue prior to 1.0.5, but they appear to be migrating away from the API used by this libary, thus no longer including the API in the precompiled binaries. 

That's why you'd get errors like:

> libraries/esp32-ps3-master/ps3_gap.c.o:(.literal.ps3_gap_init_service+0x1c): undefined reference to GAP_ConnOpen' 

In the mean time, I have resolved this issue **_for now_** in version 1.1.0 of this library, by moving back to Espressif's lower-level Bluetooth APIs (which are still compiled for Arduino at the time of writing).  Until they have officially released 1.0.5, they could still disable those APIs as well, so the library might in the end still not be compatible.

Espressif has been working hard on introducing a new API stack for using HID profiles, but it is not mature enough to be used in this library. Until they polish up their new API, I kinda have to do the splits with this library, with no guarantee that I'm able to keep it compatible with ESP-IDF.

hreintke, 2021-01-26:
Thanks for the info, clear what is the status now.

In my words : 
When the (future) Arduino API, together with the exposed ESP-IDF API in Arduino, is not enough to provide the necessary functionality you cannot support the Arduino version.

Do you (expect) then to still support the ESP-IDF version of the library ?
If so, I could move my project to ESP-IDF or compile a non-standard bluetooth library and move that to my arduino project.