jvpernis/esp32-ps3 #30: PS3_Controller_Host - help

Naruto128, 2021-02-05:
Hello, thank you very much for the bookstore.

i'm trying to get my jostick ps3 to work, but i can't.

It compiles well but it just doesn't connect, the lights don't stop flashing, the strange thing is that if I reset the esp32, the control in a few seconds turns off. that is, it kind of connects partially, but it never tells me that it has successfully connected.

I have tried it to jostick with my pc, with cable and saw bluetooth to play games and it works correctly

I'm using
ide arduino 1.8.10
ide esp32 1.0.1

PS3_Controller_Host, I have tried all versions.

Naruto128, 2021-02-07:
have found this library but it is for ps4 jostick, but it is very similar to the ps3 one. I miss it that I tried it with my ps3 control, and it shows me that it has connected, but it does not throw me any of the control status.

https://github.com/aed3/PS4-esp32

Greeka67, 2021-03-03:
You have chosen an example that only shows the controller's MAC address. You need to choose another example and write the address you received in it. Attached an example with your address, should work right away. 

[Ps3Demo.zip](https://github.com/jvpernis/esp32-ps3/files/6074424/Ps3Demo.zip)

Naruto128, 2021-03-03:
> 
> 
> You have chosen an example that only shows the controller's MAC address. You need to choose another example and write the address you received in it. Attached an example with your address, should work right away.
> 
> [Ps3Demo.zip](https://github.com/jvpernis/esp32-ps3/files/6074424/Ps3Demo.zip)

Hello. it just doesn't connect