jvpernis/esp32-ps3 #31: Reconnecting to esp32 does not work.

Greeka67, 2021-03-03:
Hello. Thanks a lot for your work!
In version 1.1.0 of the library, the ability to reconnect the controller has disappeared. The previous version had such an opportunity.
The situation is this: if you connect the controller and do not use it for about 8 minutes, it will "fall asleep", it will not connect to esp32 again.

How can I contact you for cooperation?

zusry, 2021-03-04:
I am currently having a problem with the computer.I am about to create an
app that will change the way all players play games on consoles. I will
need help.I will use c ++ and phytom.I will not author the control code. I
can't talk about the app right now because I suspect that I'm being
monitored by a hacker in the country where I am. my cell phone location
appears in several states and I have not traveled to any of these states.
I'm in social isolation, I rarely go out on the street.

zusry, 2021-03-04:
YouTube has a function that adds subtitles in other languages.

zusry, 2021-03-04:
the subtitle function is in the right corner above where the video appears.
I don't know how to use email and the first time I send a message.

Dainiakas, 2021-06-30:
> Hello. Thanks a lot for your work!
> In version 1.1.0 of the library, the ability to reconnect the controller has disappeared. The previous version had such an opportunity.
> The situation is this: if [...]

I have similar problem with disconnect - reconnect.
if I walk a bit further from ESP32 it disconnects, and there are two problems with it - you cannot reconnect (unless reset esp32) and in software it's not detecting that it's disconnected!
 Ps3.attachOnDisconnect() - does not trigger 
and Ps3.isConnected() - does not change to false

cannot handle the disconnect :(

wzli, 2021-07-01:
Yes same, tested on ESPIDF v4.3.

wzli, 2021-07-01:
I discovered that if the reconnect failed, holding the PS button to turn off the controller, then try again and it will work on the second time.

mind2reality, 2021-07-07:
Hi,

unfortunately I have the same Problem.
I am using a AZ-Delivery ESP32.
Everything is working fine, except the disconnect/reconnect.
I wanted to implement a Failsave method , but can't manage to catch the disconnection.