jvpernis/esp32-ps3 #34: PS3 controller connecting but reporting the error code: 0x02

aguaviva, 2021-03-30:
I can't seem to get data from the controller. This is a summary of the BT transactions:

1. The PS3 controller connects to the esp32 when I press the PS button
2. The esp32 sends the "sony_set_operational_bt" ("0x53, 0xf4, 0x42, 0x03, 0x00, 0x00")
3. The PS3 controller replies with a 1 byte l2cap data packet, just containing a "2", this might indicate an error as per this code http://www.pabr.org/sixlinux/patch-hidd-3.19-pabr3, see the code below "Wait for ack from device to ensure compliance with HIDP 7.9.1" expecting a 0.

Here are the details of the transaction, any help is more than welcome :)
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
esp32 BT address: 80:7D:3A:B9:BA:B2 <- This is the mac of my esp32, and it is stored in the controller

I (582) PS3_SPP: ps3_spp_init begin
I (583) PS3_SPP: ps3_spp_init end

I (583) PS3_L2CAP: [ps3_l2cap_init_services] begin

    I (1250) PS3_L2CAP: [ps3_l2cap_init_service] begin
        I (1248) PS3_SPP: ps3_spp_callback begin event 0
        I (1254) PS3_L2CAP: [ps3_l2cap_init_service] Service PS3-HIDC Initialized
        I (1259) PS3_SPP: ESP_SPP_INIT_EVT
    I (1265) PS3_L2CAP: [ps3_l2cap_init_service] end

    I (1268) PS3_SPP: esp_bt_gap_set_scan_mode

    I (1273) PS3_L2CAP: [ps3_l2cap_init_service] begin
        I (1281) PS3_L2CAP: [ps3_l2cap_init_service] Service PS3-HIDI Initialized
            I (1282) PS3_SPP: ps3_spp_callback begin event 28
    I (1287) PS3_L2CAP: [ps3_l2cap_init_service] end

I (1296) PS3_L2CAP: [ps3_l2cap_init_services] end

OK
Ready.
[I][Ps3ToBle.ino:250] setup(): TAG

************* Here I press the PS button *************

I (11482) PS3_L2CAP: [ps3_l2cap_connect_ind_cback] bd_addr: 
  l2cap_cid: 0x40
  psm: 17
  id: 1

I (11496) PS3_L2CAP: [ps3_l2cap_config_ind_cback] l2cap_cid: 0x40
  p_cfg->result: 0
  p_cfg->mtu_present: 0
  p_cfg->mtu: 0

I (11498) PS3_L2CAP: [ps3_l2cap_config_cfm_cback] l2cap_cid: 0x40
  p_cfg->result: 0

I (11524) PS3_L2CAP: [ps3_l2cap_connect_ind_cback] bd_addr: 
  l2cap_cid: 0x41
  psm: 19
  id: 3

I (11551) PS3_L2CAP: [ps3_l2cap_config_ind_cback] l2cap_cid: 0x41
  p_cfg->result: 0
  p_cfg->mtu_present: 0
  p_cfg->mtu: 0

I (11553) PS3_L2CAP: [ps3_l2cap_config_cfm_cback] l2cap_cid: 0x41
  p_cfg->result: 0

I (11559) PS3_L2CAP: ps3Enable begin
I (11562) PS3_L2CAP: [ps3_l2cap_send_hid] begin
I (11567) PS3_L2CAP: [ps3_l2cap_send_hid] len 6, event 13, offset 49188, 53, f4, 42, 03, 00, 00
I (11576) PS3_L2CAP: [ps3_l2cap_send_hid] sending command: success
I (11581) PS3_L2CAP: [ps3_l2cap_send_hid] end
I (11585) PS3_L2CAP: ps3Enable end

I (11604) PS3_L2CAP: [ps3_l2cap_data_ind_cback] begin
I (11604) PS3_L2CAP: [ps3_l2cap_data_ind_cback] packet len 1, data: 02
I (11604) PS3_L2CAP: [ps3_l2cap_data_ind_cback] end
```

Any help with this on how to debug it is more than welcome :) 

Oh! And thanks for this great project!!

Naruto128, 2021-04-05:
Hello, I also have the problem. apparently the problem is with Joystick clones. I solved it but it is not a direct way but it could be a solution with some advantages, such as the wifi, greater range of reach. 

https://subefotos.com/ver/?fbccec11ff9f8dad65eaf00a0bdb7a98o.jpg

aguaviva, 2021-04-05:
Could you please explain how did you solve it?

Naruto128, 2021-04-06:
First you must know if your joystick is compatible with your phone, then my application enters into action that makes the link to the esp32. but my application has a cost 

Sixaxis Compatibility Checker
https://play.google.com/store/apps/details?id=com.dancingpixelstudios.sixaxiscompatibilitychecker&hl=es_AR

Naruto128, 2021-04-06:
https://www.mediafire.com/view/zh9lyhyn67bckbl/Sin_t%25C3%25ADtulo.png/file
https://www.mediafire.com/view/sa6jlacs11nu2i0/Screenshot_2021-04-05-21-39-13.png/file
https://www.mediafire.com/view/42ivvlkg55ojbxo/Screenshot_2021-04-05-21-39-04.png/file
https://www.mediafire.com/view/6jeix7032thqtsr/Screenshot_2021-04-05-21-38-38.png/file

aguaviva, 2021-04-09:
I found the solultion, my controller is a Gasia clone, you need to use this code to get the controller to stream its state

```cpp
void ps3Enable()
{
    uint16_t len = sizeof(hid_cmd_payload_ps3_enable);
    hid_cmd_t hid_cmd;

#ifdef GASIA
//*****************************Begin hack***************************************
	static const uint8_t hid_cmd_payload_gasia_enable[] = {
			//0xA2,        0x01,
			0x00, 0x00, 0x00, 0x00, 0x00,   // rumble values [0x00, right-timeout, right-force, left-timeout, left-force]
			0x00, 0x00, 0x00, 0x00, 0x02,   // 0x02=LED1 .. 0x10=LED4
			0xff, 0x27, 0x10, 0x00, 0x32,   // LED 4
			0xff, 0x27, 0x10, 0x00, 0x32,   // LED 3
			0xff, 0x27, 0x10, 0x00, 0x32,   // LED 2
			0xff, 0x27, 0x10, 0x00, 0x32,   // LED 1
			0x00, 0x00, 0x00, 0x00, 0x00,
			0x00, 0x00, 0x00, 0x00, 0x00,
			0x00, 0x00, 0x00, 0x00, 0x00,
			0x00, 0x00, 0x00
		};
    uint16_t len = sizeof(hid_cmd_payload_gasia_enable);
    hid_cmd_t hid_cmd;
    hid_cmd.code = 0xA2; 
    hid_cmd.identifier = 0x01;
    memcpy( hid_cmd.data, hid_cmd_payload_gasia_enable, len);
//*****************************End hack***************************************
#else
    hid_cmd.code = hid_cmd_code_set_report | hid_cmd_code_type_feature;
    hid_cmd.identifier = hid_cmd_identifier_ps3_enable;
    memcpy( hid_cmd.data, hid_cmd_payload_ps3_enable, len);
#endif
    ps3_l2cap_send_hid( &hid_cmd, len );
}
```

Naruto128, 2021-04-10:
hello, it works !!! it connects and throws me the information all correct. but the LEDs start to flash, they change places. It's like this for a while, and then the control is turned off

aguaviva, 2021-04-10:
For more info see:
https://github.com/falkTX/qtsixa/blob/c2f4ed9ee5646aefa68fb44fa14eda0460f02e66/sixad/shared.cpp#L272

and search for GASIA_GAMEPAD_HACKS in that repo to see how to set up the rumble and lights...

ashokpkumar, 2024-05-01:
If someone still needs it. this might be also due to the esp32. 
the same code giving problem in old esp32 works fine in new esp32 board. 
the old esp32 board requuires to press the bootsel button while uploading while the old one does not.
the absence of the capacitor also might be an issue.