jvpernis/esp32-ps3 #39: Controller Disconnect handling

mind2reality, 2021-07-08:
Hi,
Thank you for this fantastic library :)
I have allready commented into another open issue, but got no Response. 
So, here I tell my Problem again ;)

I am using a ps3 Navigation Controller with the ESP32 "wroom" from AZdelivery.

Everything is working fine and I am very happy implementing all those Events easiely.

But, I have no chance to catch any disconnect event.
I wanted to implement some failsave code to my rc System, unfortunately that isn't possible for me at the Moment. 

I would be very happy for any help at this Point.

Best regards

trymha, 2021-08-06:
Hello. I have the same issue. Were you able to solve it?

mind2reality, 2021-08-06:
unfortunately I haven't found a solution yet ...

GustavoDRL, 2021-10-12:
I had this problem too. Have you found a solution?

TD88robot, 2022-06-02:
> there is a fork of this project where someone else added the disconnection handling: I tested it - It works well. https://github.com/RomanLut/esp32-ps3/tree/1.0.0_with_reconnect_fixed

Nice tip! I also got this working with the suggested project. Also install an older version of the esp32 board package. Version 1.0.4 will avoid compiler errors

simonliu009, 2022-09-13:
> there is a fork of this project where someone else added the disconnection handling: I tested it - It works well. https://github.com/RomanLut/esp32-ps3/tree/1.0.0_with_reconnect_fixed

This fork won't compile on ESP-IDF v4.4.2