jvpernis/esp32-ps3 #40: ESP32 to Ps3 controllers will not connect, followed all known steps, can you help?

alexinstrell, 2021-09-17:
Hi
Firstly, I'm very grateful for you putting this code up online, and excited to try and use it on my projects. But I have been trying to get this to work for about a week now with no luck. I have four ESP32 development boards, all are NodeMCU ESP32S Devkit V1.
I have two official Sony PS3 Controllers, one is a Sixaxis the other a Sixaxis Dual Shock 3 (both are over 10 years old, but can still pair with the games console itself). I don't know if country is relevant to the model of controller, but I'm in the UK.
Finally, I am using the Arduino IDE 1.8.15, with the Espressif ESP32 Board Manager installed, and your Ps3Controller library.

I have obtained the bluetooth MAC addresses for all the ESP32s using 'Ps3Address'. I have used the Sixaxis pairing tool to discover the stored MAC in the controllers, and then change it to the ESP32 BT MAC (and checked that it stays in the memory). 

I have then tried to use the Ps3Connect sketch, the only change I have made is to add in my board's BT MAC. Here is the code:
____

#include <Ps3Controller.h>

void setup()
{
    Serial.begin(115200);
    Ps3.begin("7c:9e:bd:48:bf:02");
    Serial.println("Ready.");
}

void loop()
{
  if (Ps3.isConnected()){
    Serial.println("Connected!");
  }

  delay(3000);
}

__

It compiles and uploads fine, I open the serial monitor, reset the ESP32 using the EN button, and it reads:
___
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1100
ho 0 tail 12 room 4
load:0x40078000,len:12308
load:0x40080400,len:3076
entry 0x400805ec
Ready.

___

I then push the connect button (central one!) on the Ps3 (have tried quick push and holding down), the 4 red LEDS on the back flash quickly for around a minute, but nothing else happens. Nothing more in the serial monitor. Have also tried resetting the ESP32 whilst controller LEDS are flashing. Have tried every combination of my four NodeMCUs with both controllers. Have reset the original PS3 console MAC into the controllers, and they pair again. I have even tried uninstalling then reinstalling the ESP32 board manager and the PS3Controller library, and tried regressing to previous library versions.

 Have tried looking for alternative code on the internet but nothing else works. 

Can you help in any way? Is this something to do with the espressif IDF? Does my development board not work with PS3 BT for some reason? I cannot think what else to try.

Many thanks for any assistance you can offer.

Kind regards

Alex

alexinstrell, 2021-10-08:
Hi 86Leonidas2, thanks so much for having a look at this. I've given it a go this morning, following your instructions, and with a few variations of ESP32 and Ps3 Controller (and varying which MAC address is put into the controller and the code, trying the original Ps3 Console MAC, the BT MAC from the ESP32 and even the '01: (to) 06' format as used in the demo sketch, having used sixaxis to change it. But it gets errors in compiling. Here's the error message (alas I don't know enough to interpret this, it seems there is something about 'const char' vs 'char' going on, plus other issues...

alexinstrell, 2021-10-08:
Below is the error message that I got in compilation, when I tried to upload as described.
The MAC address inserted is the BT MAC of that particular ESP32, I used sixaxis pairing tool to put this MAC into the Ps3 Controller also.

In case the format of the MAC is relevant:
7c:9e:bd:48:bf:02

(I also tried this with the original MAC address from the controller,
00:1a:80:d1:03:89

that relates to the Ps3 Games console, putting that in the code and controller). Doesn’t work either atm.)

Error message (verbatim except I have redacted the full file path where the library sits under and replaced with text ‘XXXXFILEPATH’; I did check the library was in the right place, and I’ve not had a ‘library not found’ error):
--

Arduino: 1.8.11 (Windows 10), Board: "Node32s, Default, 80MHz, 921600, None"
XXXXFILEPATHXXXX\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c: In function 'ps3_l2cap_init_services':
XXXXFILEPATHXXXX\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:77:29: warning: passing argument 1 of 'ps3_l2cap_init_service' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     ps3_l2cap_init_service( "PS3-HIDC", BT_PSM_HIDC, BTM_SEC_SERVICE_FIRST_EMPTY   );
                             ^
XXXXFILEPATHXXXX\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:27:13: note: expected 'char *' but argument is of type 'const char *'
 static void ps3_l2cap_init_service( char *name, uint16_t psm, uint8_t security_id);
             ^
XXXXFILEPATHXXXX\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:78:29: warning: passing argument 1 of 'ps3_l2cap_init_service' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     ps3_l2cap_init_service( "PS3-HIDI", BT_PSM_HIDI, BTM_SEC_SERVICE_FIRST_EMPTY+1 );
                             ^

XXXXFILEPATHXXXX\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:27:13: note: expected 'char *' but argument is of type 'const char *'
 static void ps3_l2cap_init_service( char *name, uint16_t psm, uint8_t security_id);
             ^
XXXXFILEPATHXXXX\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c: In function 'ps3_l2cap_deinit_services':
XXXXFILEPATHXXXX\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:92:31: warning: passing argument 1 of 'ps3_l2cap_deinit_service' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     ps3_l2cap_deinit_service( "PS3-HIDC", BT_PSM_HIDC );
                               ^
XXXXFILEPATHXXXX\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:28:13: note: expected 'char *' but argument is of type 'const char *'
 static void ps3_l2cap_deinit_service( char *name, uint16_t psm );
             ^
XXXXFILEPATHXXXX\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:93:31: warning: passing argument 1 of 'ps3_l2cap_deinit_service' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     ps3_l2cap_deinit_service( "PS3-HIDI", BT_PSM_HIDI );
                               ^
XXXXFILEPATHXXXX\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:28:13: note: expected 'char *' but argument is of type 'const char *'
 static void ps3_l2cap_deinit_service( char *name, uint16_t psm );
             ^
XXXXFILEPATHXXXX\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c: At top level:
cc1.exe: warning: unrecognized command line option '-Wno-frame-address'
c:/users/kelly/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find crt1-sim.o: No such file or directory
c:/users/kelly/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find _vectors.o: No such file or directory
collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Node32s.
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

alexinstrell, 2021-10-08:
If you or anyone can unpick that I'd be really grateful! Thanks

parvaizahmad, 2021-10-15:
@alexinstrell thanks for response. By the way I have figured out the solution but was not able to trace the exact problem. If you want to work with PS3 controller you have to downgrade you Arduino IDE, ESP32 library as wall as PS3 Host controller library. Working combination is below:
Arduino IDE: Version 1.8.11
ESP32 library : Version 1.0.2
PS3 Host Controller Library: Version 1.0.0
If you have already installed an arduino ide uninstall and make sure that all the temp files as well as caches are removed related to arduino libs and IDE, make a fresh installation of above mentioned components and it will work. 

+ Do let me know about PS4 with esp32.

parvaizahmad, 2021-10-15:
@alexinstrell did you removed the caches of Arduino IDE?  You can use https://dl.espressif.com/dl/package_esp32_index.json link to install the board, well on my end I am able to install all the things without any error, use steps mentioned here to remove arduino ide, 
https://support.arduino.cc/hc/en-us/articles/360021325733-How-to-do-a-complete-uninstall-of-the-Arduino-IDE 
By the way my first name is Parvaiz

parvaizahmad, 2021-10-15:
@alexinstrell contact me on my email id I may be able to help you through call.

Sriram510, 2022-06-01:
> Hi Parvaiz, thanks so much for this and for the new video on Youtube, I've also added this comment there and subscribed. I went through your advice on removal of old (hidden) cache files when I uninstalled the current IDE ready to reload the previous version, and this seems to be the thing that made the different from previous unsuccessful attempts. Didn't try 1.8.11 as per suggestion on GitHub, as I saw you used 1.8.15 in this video, and that has worked. Do we know whether it still works with latest IDE 1.8.16? For anyone else struggling with this, I also noticed that when reinstalling the IDE, the file location for your sketchbook will revert back to the default 'Arduino' folder; if, like me, you had set a custom folder previously, don't forget when you reinstall the IDE, to set it to the folder you want and make sure you have only libraries you want in there. I wondered why on reinstall it said the Ps3Controller host library (current version) was already installed (despite me having removed the file)...that was the reason, it reverted to looking in the old sketchbook file location, which I had forgotten about and still contained the library file. Having 'cleaned up' and got rid of all old files, the way was clear to install these old versions, and it all worked.
> 
> Thank you for this! I do have a follow on question; should I want to now use this code to control a device (in this case a rover), but combine it with other code (sensors etc) I need for the whole machine, are we going to run into problems, as we are now using older 'unsupported' IDE, board manager and library? I know the above solution can work in isolation, it's a question of can we/how do we make this work 'permanently' alongside other functions.
> 
> I'm now going to try and collate together good code for driving a rover using these control inputs (probably single stick as you have done). Mine is 4wd with 4 gear motors, left and right motors 'tied' together, with a TB6612FNG motor driver. That leaves all the other controls free for robot arms, sensors and so on.

It still doesn't work for me. do you know if how the code can be written in the IDE to connect and read data?

vandjac, 2022-06-13:
I followed all the steps in Parvaiz's video, and it still didn't work for me. I am using a knockoff Chinese controller. Do you have to use a genuine Sony PS3 controller?

parvaizahmad, 2022-06-17:
@Erkki507 are you connecting ground properly? Can you check output voltages of L298 driver at 5 volts terminal?

parvaizahmad, 2022-06-17:
@vandjac it should work fine, may be there is some issue with your controller or esp32 board,

Erkki507, 2022-07-28:
Hi, 
sorry I have had other projects so I haven't been able to work with this project. I have with another ps3 controller and checked voltages and connected everything multiple times but still only way to get ESP32 and controlled connected is use own powersupply for ESP32 trought micro usb port. I have ordered another ESP32 so we will see if it works right way or is there some problems with my wiring or code.

ypatidisioannis, 2023-02-18:
Hello there! 

I am having an issue trying to connect PS3 OEM controllers with an ESP32 but I can't. Does anyone have a solution with that?

Actually what I see is that the sixaxistool can't write the address of the master in BT . I open the sixaxistool, I connect the OEM controller and I push the update button, then I unplug the controlles from the USB and when I plug it again it doesn't show me the address I wrote. 

Any ideas why?

Thank you!

Muntaharul, 2023-06-08:
> Hello there!
>
> I am having an issue trying to connect PS3 OEM controllers with an ESP32 but I can't. Does anyone have a solution with that?
>
> Actually what I see is that the sixaxistool can't write [...]

hi you you solv the problem

Muntaharul, 2023-06-08:
> I followed all the steps in Parvaiz's video, and it still didn't work for me. I am using a knockoff Chinese controller. Do you have to use a genuine Sony PS3 controller?

have you find the solution ps3 not connect to esp32

Jangidyogesh12, 2025-01-08:
why my i am not getting my mac address on the six axis pair tool because when i open it. It is unable to serch mac address of my ps3 cnotroller. It is showing that unable to find device even if i have connected it .
Can anyone please help me with this.