jvpernis/esp32-ps3 #45: Led lights and Rubble not working

ORANGEID, 2022-01-19:
Hi There

First of thanks for this fantastic library works really well on the getting feedback from the the button just what i needed

Unfortunately i am not have much luck with feeding back to the controller

I am running on a esp32, coding in arduino and using what assume is a knock off ps3 controller

i am running the following command

// To set led 1 to light up
Ps3.setPlayer(1);
Verify's fine but doesn't seem to make any changes on the ps3, also give no errors message

// to ruble controller
Ps3.setRumble(100.0, 1000);
Verify's fine
Report following error message "E (544498) PS3_L2CAP: [ps3_l2cap_send_hid] sending command: failed"

Any help would greatly appreciated

James

kristian-abood, 2022-02-06:
Hey James, 

I'm having the exact same issue with my knock off ps3 controller. 

Any help would for sure be appreciated! 

Thanks,
Kristian

agent6, 2022-07-07:
I am having the same issue.