jvpernis/esp32-ps3 #49: Does not work on platform-espressif32 4.0.0 or more.

asukiaaa, 2022-05-24:
This library does not work on PlatformIO with using platform-espressif32 [4.0.0](https://github.com/platformio/platform-espressif32/releases/tag/v4.0.0) (arduino v2.0.0) or more(4.1.0, 4.2.0 and 4.2.0) but works with [3.5.0](https://github.com/platformio/platform-espressif32/releases/tag/v3.5.0).
Is there any way to work on 4.0.0 or more?

Thank you.

julian-weinert, 2022-05-28:
Hi @asukiaaa! I've had to modify the CMakeList.txt for the ps3 component
There is also a fork and an open pull request for 4.0 support, you may simply use that.

I had connection trouble, so maybe it's some incompatibility thing, but I've not had any look searching / reaching out.

```
idf_component_register(SRCS "src/ps3.c"
                            "src/ps3_spp.c"
                            "src/ps3_parser.c"
                            "src/ps3_l2cap.c"
                        REQUIRES nvs_flash bt
                        PRIV_REQUIRES bt
                        INCLUDE_DIRS src/include
                        PRIV_INCLUDE_DIRS   ${IDF_PATH}/components/bt/common/include/
                                            ${IDF_PATH}/components/bt/host/bluedroid/common/include/
                                            ${IDF_PATH}/components/bt/host/bluedroid/stack/include)
```

asukiaaa, 2022-05-29:
Thank you for the response.

> I've had to modify the CMakeList.txt for the ps3 component
> There is also a fork and an open pull request for 4.0 support, you may simply use that.

Can you show some urls for them?

julian-weinert, 2022-05-31:
It might be worth mentioning that the fork of rrooding uses a pre-release bluetooth GAP API that is not supported by 4.0 anymore.
I would recommend using this original fork and add the `CMakeList.txt` as I have it above.
There is another change in rroodings fork, in the `src/ps3_spp.c:150`. It appears that ESP-IDF now supports the use of `ESP_SPP_SEC_AUTHORIZE` instead of `ESP_SPP_SEC_NONE` but I'm not sure how important it is.

No matter what I do I can't seem to be able to get it connecting. It might be necessary to go over the project again and maybe convert it to the new official gap API if it supports the needed functionality.