jvpernis/esp32-ps3 #53: Alternative for SixaxisPairTool and sixaxispairer

biemster, 2023-01-10:
The link to SixaxisPairTool seems to have gone all scammy, and sixaxispairer has a dependency that is no longer available at the link provided in that repo. I've found and (ancient) alternative here: http://www.pabr.org/sixlinux, for people that might be searching like I did. It might be even an idea to add `sixpair.c` to this repo, as to not lose it.

anatol-grabowski, 2023-02-15:
I was able to compile sixaxispairer from this repo [https://github.com/crazyquark/sixaxispairer] after installing "hidapi" by pacman on Manjaro.

biemster, 2023-02-15:
I see now that there is a `libhidapi` on my Ubuntu machine as well, so I could have tried that..
Anyway the `sixpair.c` is still a good alternative to consider.