jvpernis/esp32-ps3 #54: Cannot connect controller to esp32

nnnnaufalir, 2023-01-17:
i follow all step use this library, but my controller could not connect to esp32, i was change my MAC addr, and read all issues in this repo, but not solve my problem, can anyone help this issues?

anatol-grabowski, 2023-02-15:
I have two controllers, both linked to the same MAC. The original one does NOT work for me (doesn't pair with esp32). A clone branded "Esperanza" works fine.

nnnnaufalir, 2023-03-08:
> I have two controllers, both linked to the same MAC. The original one does NOT work for me (doesn't pair with esp32). A clone branded "Esperanza" works fine.

my problem solve with change my esp32 IC, previously i use IC ESP32 Wroom usual, then i change my IC of ESP32 with ESP32D, and the stick is connect without any problems. What series of IC esp32 you use?

MortadhaDAHMANI, 2023-06-05:
my problem solve with change my esp32 IC, previously i use IC ESP32 Wroom usual, then i change my IC of ESP32 with ESP32D, and the stick is connect without any problems. What series of IC esp32 you use?

MortadhaDAHMANI, 2023-06-05:
I having the same issue.

anatol-grabowski, 2023-06-26:
@nnnnaufalir Mine says "ESP32-WROOM-32D" on top of it.

Oleoleo2, 2025-03-29:
Same problem here, and I'm running WROOM-32D