jvpernis/esp32-ps3 #57: Problem with New Arduino Version

ypatidisioannis, 2023-03-09:
Hello guys!

I ave installed the latest version of Arduino to support the ESP32 DEV S3 board, and when I compile my code I get many erros.

Do you have any work around?

Thank you in advance!

gwarble, 2023-04-08:
Library looks awesome, many thanks I also hope to get it working with the latest Arduino and ESP32S3, makes it to "linking everything together" then I get these errors:

```
temp\path\to\libraries\PS3_Controller_Host\ps3_l2cap.c.o:(.literal.ps3_l2cap_config_ind_cback+0x0): undefined reference to 'L2CA_ConfigRsp'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_l2cap.c.o:(.literal.ps3_l2cap_connect_ind_cback+0x4): undefined reference to 'L2CA_ErtmConnectRsp'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_l2cap.c.o:(.literal.ps3_l2cap_connect_ind_cback+0x8): undefined reference to 'L2CA_ConfigReq'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_l2cap.c.o:(.literal.ps3_l2cap_init_service+0x14): undefined reference to 'L2CA_Register'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_l2cap.c.o:(.literal.ps3_l2cap_send_hid+0x14): undefined reference to 'L2CA_DataWrite'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_l2cap.c.o: in function 'ps3_l2cap_config_ind_cback':
c:\Users\User\Documents\Arduino\libraries\PS3_Controller_Host\src/ps3_l2cap.c:263: undefined reference to 'L2CA_ConfigRsp'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_l2cap.c.o: in function 'ps3_l2cap_connect_ind_cback':
c:\Users\User\Documents\Arduino\libraries\PS3_Controller_Host\src/ps3_l2cap.c:197: undefined reference to 'L2CA_ErtmConnectRsp'
path/to/ld.exe: c:\Users\User\Documents\Arduino\libraries\PS3_Controller_Host\src/ps3_l2cap.c:200: undefined reference to 'L2CA_ErtmConnectRsp'
path/to/ld.exe: c:\Users\User\Documents\Arduino\libraries\PS3_Controller_Host\src/ps3_l2cap.c:203: undefined reference to 'L2CA_ConfigReq'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_l2cap.c.o: in function 'ps3_l2cap_init_service':
c:\Users\User\Documents\Arduino\libraries\PS3_Controller_Host\src/ps3_l2cap.c:152: undefined reference to 'L2CA_Register'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_l2cap.c.o: in function 'ps3_l2cap_send_hid':
c:\Users\User\Documents\Arduino\libraries\PS3_Controller_Host\src/ps3_l2cap.c:122: undefined reference to 'L2CA_DataWrite'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_spp.c.o:(.literal.ps3_spp_callback+0x8): undefined reference to 'esp_bt_gap_set_scan_mode'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_spp.c.o:(.literal.ps3_spp_callback+0xc): undefined reference to 'esp_spp_start_srv'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_spp.c.o:(.literal.ps3_spp_init+0x14): undefined reference to 'esp_spp_register_callback'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_spp.c.o:(.literal.ps3_spp_init+0x18): undefined reference to 'esp_spp_init'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_spp.c.o: in function 'ps3_spp_callback':
c:\Users\User\Documents\Arduino\libraries\PS3_Controller_Host\src/ps3_spp.c:145: undefined reference to 'esp_bt_gap_set_scan_mode'
path/to/ld.exe: c:\Users\User\Documents\Arduino\libraries\PS3_Controller_Host\src/ps3_spp.c:150: undefined reference to 'esp_spp_start_srv'
path/to/ld.exe: temp\path\to\libraries\PS3_Controller_Host\ps3_spp.c.o: in function 'ps3_spp_init':
c:\Users\User\Documents\Arduino\libraries\PS3_Controller_Host\src/ps3_spp.c:68: undefined reference to 'esp_spp_register_callback'
path/to/ld.exe: c:\Users\User\Documents\Arduino\libraries\PS3_Controller_Host\src/ps3_spp.c:77: undefined reference to 'esp_spp_init'
collect2.exe: error: ld returned 1 exit status

Using library PS3 Controller Host at version 1.1.0 in folder: C:\Users\User\Documents\Arduino\libraries\PS3_Controller_Host 
exit status 1

Compilation error: exit status 1
```

where these paths are shortened for viewing in post:
`C:\Users\User\AppData\Local\Temp\arduino\sketches\8FC144A5356F9DCC7A6F2FBC9A08EBE8\libraries\PS3_Controller_Host\ps3_l2cap.c.o`
`c:/users/user/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe`

in case this helps anyone diagnose the issue...  I'm not an expert but I tried it on two independent setups, latest v1.1.0 and v1.0.0, but have not tried an older than current 2.0.7 esp32 espressif board manager

`#include <Ps3Controller.h>` alone compiles fine but only adding `Ps3.begin("01:02:03:04:05:06");` causes the above errors

ypatidisioannis, 2023-04-19:
I have seen that not all versions of the ESP32 are good

closed 2023-04-19 by ypatidisioannis