jvpernis/esp32-ps3 #60: Can't connect

shan-mcarthur, 2023-05-11:
I have a new PS3 controller.  I have no PS3 system as I bought this controller just for my project.  It is not already paired to any PS3 system.  The Sixaxis Pair tool can't seem to connect to the controller when I have it plugged into my PC using USB.  The controller is working on my PC because when I click buttons it opens up xbox control center and I can navigate around using the joysticks and buttons on the controller.  I have obtained my bluetooth MAC address using your sample code and am using the connect sketch with that mac address in the begin() method.  There are no errors but it never does complete the connect.  How can I get this connected to my SparkFun ESP32 WROOM Smart Thing Plus C board?

Ja-bots, 2023-12-19:
Were you able to solve your problem? I have the same problem

Abudi-seid10, 2024-03-13:
hello guys 
im having the same problem 
please contact me at abduseid28@gmail.com if u find any solutions

butcher51, 2024-05-18:
I have the same problem, it seems impossible to get the mac address :/

Oleoleo2, 2025-03-29:
Had the same problem, just using sixaxispairer (also tried sixaxispairertool on Windows). I got the Mac addresse out of it by holding the ps3 button down, while running the sixaxispairer command (try af few times), don't know if that works in sixaxispairertool as well. Though I can still not get it to connect to the esp32. It's an Aliexpress cheap ps3 controller, don't know if the problem is related to this.