jvpernis/esp32-ps3 #64: Compiling issues

JerryFTyson, 2023-11-06:
Is this what you mean as an issue with broken tools:
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_l2cap.c.o:(.literal.ps3_l2cap_config_ind_cback+0x0): undefined reference to `L2CA_ConfigRsp'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_l2cap.c.o:(.literal.ps3_l2cap_connect_ind_cback+0x4): undefined reference to `L2CA_ErtmConnectRsp'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_l2cap.c.o:(.literal.ps3_l2cap_connect_ind_cback+0x8): undefined reference to `L2CA_ConfigReq'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_l2cap.c.o:(.literal.ps3_l2cap_init_service+0x14): undefined reference to `L2CA_Register'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_l2cap.c.o:(.literal.ps3_l2cap_send_hid+0x14): undefined reference to `L2CA_DataWrite'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_l2cap.c.o: in function `ps3_l2cap_config_ind_cback':
/Users/Home/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3_l2cap.c:263: undefined reference to `L2CA_ConfigRsp'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_l2cap.c.o: in function `ps3_l2cap_connect_ind_cback':
/Users/Home/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3_l2cap.c:197: undefined reference to `L2CA_ErtmConnectRsp'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/Home/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3_l2cap.c:200: undefined reference to `L2CA_ErtmConnectRsp'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/Home/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3_l2cap.c:203: undefined reference to `L2CA_ConfigReq'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_l2cap.c.o: in function `ps3_l2cap_init_service':
/Users/Home/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3_l2cap.c:152: undefined reference to `L2CA_Register'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_l2cap.c.o: in function `ps3_l2cap_send_hid':
/Users/Home/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3_l2cap.c:122: undefined reference to `L2CA_DataWrite'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_spp.c.o:(.literal.ps3_spp_callback+0x8): undefined reference to `esp_bt_gap_set_scan_mode'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_spp.c.o:(.literal.ps3_spp_callback+0xc): undefined reference to `esp_spp_start_srv'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_spp.c.o:(.literal.ps3_spp_init+0x14): undefined reference to `esp_spp_register_callback'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_spp.c.o:(.literal.ps3_spp_init+0x18): undefined reference to `esp_spp_init'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_spp.c.o: in function `ps3_spp_callback':
/Users/Home/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3_spp.c:145: undefined reference to `esp_bt_gap_set_scan_mode'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/Home/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3_spp.c:150: undefined reference to `esp_spp_start_srv'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/cn8cjqs50qx6fq0fcc72zcgh0000gn/T/arduino/sketches/067475F8CC8E42CC39C316CF0743CC94/libraries/PS3_Controller_Host/ps3_spp.c.o: in function `ps3_spp_init':
/Users/Home/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3_spp.c:68: undefined reference to `esp_spp_register_callback'
/Users/Home/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/Home/Documents/Arduino/libraries/PS3_Controller_Host/src/ps3_spp.c:77: undefined reference to `esp_spp_init'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

matfitzsimon, 2023-11-19:
Hi, yes I’m having the same issue.

Newreggie, 2023-11-19:
I advise that you check the SPP file that it makes reference to, also I made use of the ps4 controller and it worked instantly so perhaps give that a try

matfitzsimon, 2023-11-19:
> I advise that you check the SPP file that it makes reference to, also I made use of the ps4 controller and it worked instantly so perhaps give that a try

Hi, do you mean a ps4 controller?

wannn28, 2024-05-08:
Maybe it's caused when you use ESP32S3, try using a regular ESP32