jvpernis/esp32-ps3 #78: Arduino Nano esp32-S3 + PS3 controller, "undefined reference: xxx" "exit status 1"

AstroPanda-maker, 2024-09-22:
Hello Newbie here
I was going through "DroneBot Workshop esp32 and Ps3 controll" tutorial. 

Im struggling to get any of the example sketches to work with the Arduino Nano ESP32.

Everytime i try to verify the sketch i get several red lines of "undefined reference" errors preventing the sketch from verifying. 

What i've tried so far. 
I've reinstalled Arduino IDE and reinstalled the library and reinstall all the board managers. Did not work. :(

Board Arduino Nano ESP32-S3

Arduino IDE version 2.3.2
Board managers
Arduino AVR 1.8.6
Arduino ESP32 boards 2.0.17
ESP32 3.0.4

Error Codes: 
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_l2cap.c.o:(.literal.ps3_l2cap_init_service+0x14): undefined reference to `L2CA_Register'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_l2cap.c.o:(.literal.ps3_l2cap_config_ind_cback+0x0): undefined reference to `L2CA_ConfigRsp'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_l2cap.c.o:(.literal.ps3_l2cap_connect_ind_cback+0x4): undefined reference to `L2CA_ErtmConnectRsp'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_l2cap.c.o:(.literal.ps3_l2cap_connect_ind_cback+0x8): undefined reference to `L2CA_ConfigReq'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_l2cap.c.o:(.literal.ps3_l2cap_send_hid+0x10): undefined reference to `L2CA_DataWrite'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_l2cap.c.o: in function `ps3_l2cap_init_service':
c:\Users\TechC\Documents\Arduino\Library\libraries\PS3_Controller_Host\src/ps3_l2cap.c:152: undefined reference to `L2CA_Register'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_l2cap.c.o: in function `ps3_l2cap_config_ind_cback':
c:\Users\TechC\Documents\Arduino\Library\libraries\PS3_Controller_Host\src/ps3_l2cap.c:263: undefined reference to `L2CA_ConfigRsp'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_l2cap.c.o: in function `ps3_l2cap_connect_ind_cback':
c:\Users\TechC\Documents\Arduino\Library\libraries\PS3_Controller_Host\src/ps3_l2cap.c:197: undefined reference to `L2CA_ErtmConnectRsp'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: c:\Users\TechC\Documents\Arduino\Library\libraries\PS3_Controller_Host\src/ps3_l2cap.c:200: undefined reference to `L2CA_ErtmConnectRsp'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: c:\Users\TechC\Documents\Arduino\Library\libraries\PS3_Controller_Host\src/ps3_l2cap.c:203: undefined reference to `L2CA_ConfigReq'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_l2cap.c.o: in function `ps3_l2cap_send_hid':
c:\Users\TechC\Documents\Arduino\Library\libraries\PS3_Controller_Host\src/ps3_l2cap.c:122: undefined reference to `L2CA_DataWrite'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_spp.c.o:(.literal.ps3_spp_callback+0x8): undefined reference to `esp_bt_gap_set_scan_mode'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_spp.c.o:(.literal.ps3_spp_callback+0xc): undefined reference to `esp_spp_start_srv'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_spp.c.o:(.literal.ps3_spp_init+0x14): undefined reference to `esp_spp_register_callback'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_spp.c.o:(.literal.ps3_spp_init+0x18): undefined reference to `esp_spp_init'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_spp.c.o: in function `ps3_spp_callback':
c:\Users\TechC\Documents\Arduino\Library\libraries\PS3_Controller_Host\src/ps3_spp.c:145: undefined reference to `esp_bt_gap_set_scan_mode'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: c:\Users\TechC\Documents\Arduino\Library\libraries\PS3_Controller_Host\src/ps3_spp.c:150: undefined reference to `esp_spp_start_srv'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\TechC\AppData\Local\Temp\arduino\sketches\CAB5C5B55EB4A65A1A64888E24069CEA\libraries\PS3_Controller_Host\ps3_spp.c.o: in function `ps3_spp_init':
c:\Users\TechC\Documents\Arduino\Library\libraries\PS3_Controller_Host\src/ps3_spp.c:68: undefined reference to `esp_spp_register_callback'
c:/users/techc/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: c:\Users\TechC\Documents\Arduino\Library\libraries\PS3_Controller_Host\src/ps3_spp.c:77: undefined reference to `esp_spp_init'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

CraXyOW3, 2024-09-22:
Got a similar error but for ESP32-C3.

mkvst2, 2025-01-19:
Same problem, Nano ESP32-S3

Been working on it for 48 hours, think I have tried everything to the limit of my knowledge.