jvpernis/esp32-ps3 #87: any exemple code are compiling event tthe simplest

robotdelta, 2025-04-12:
about 6 month before i had a code which is compiling succesfully of an ESP32 WROOM but now it doesn't compiling anymore. 
I can't remenber the versions of my setup but last time I compiled it succesfully was the 5/01/2025 on a "ESP32_WROOM_DA MODULE".
Now i code under arduino IDE 2.3.4 with board library ESP32 bt espressif systems version 3.2.0 on a "ESP32_WROOM_DA MODULE".

I get this error: 
`c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3.c: In function 'ps3SetBluetoothMacAddress':
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3.c:287:5: error: implicit declaration of function 'esp_base_mac_addr_set' [-Wimplicit-function-declaration]
  287 |     esp_base_mac_addr_set(base_mac);
      |     ^~~~~~~~~~~~~~~~~~~~~
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c: In function 'ps3_l2cap_init_services':
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:77:29: warning: passing argument 1 of 'ps3_l2cap_init_service' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   77 |     ps3_l2cap_init_service( "PS3-HIDC", BT_PSM_HIDC, BTM_SEC_SERVICE_FIRST_EMPTY   );
      |                             ^~~~~~~~~~
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:27:43: note: expected 'char *' but argument is of type 'const char *'
   27 | static void ps3_l2cap_init_service( char *name, uint16_t psm, uint8_t security_id);
      |                                     ~~~~~~^~~~
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:78:29: warning: passing argument 1 of 'ps3_l2cap_init_service' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   78 |     ps3_l2cap_init_service( "PS3-HIDI", BT_PSM_HIDI, BTM_SEC_SERVICE_FIRST_EMPTY+1 );
      |                             ^~~~~~~~~~
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:27:43: note: expected 'char *' but argument is of type 'const char *'
   27 | static void ps3_l2cap_init_service( char *name, uint16_t psm, uint8_t security_id);
      |                                     ~~~~~~^~~~
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c: In function 'ps3_l2cap_deinit_services':
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:92:31: warning: passing argument 1 of 'ps3_l2cap_deinit_service' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   92 |     ps3_l2cap_deinit_service( "PS3-HIDC", BT_PSM_HIDC );
      |                               ^~~~~~~~~~
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:28:45: note: expected 'char *' but argument is of type 'const char *'
   28 | static void ps3_l2cap_deinit_service( char *name, uint16_t psm );
      |                                       ~~~~~~^~~~
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:93:31: warning: passing argument 1 of 'ps3_l2cap_deinit_service' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   93 |     ps3_l2cap_deinit_service( "PS3-HIDI", BT_PSM_HIDI );
      |                               ^~~~~~~~~~
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_l2cap.c:28:45: note: expected 'char *' but argument is of type 'const char *'
   28 | static void ps3_l2cap_deinit_service( char *name, uint16_t psm );
      |                                       ~~~~~~^~~~
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_spp.c: In function 'ps3_spp_init':
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_spp.c:73:5: warning: 'esp_spp_init' is deprecated: Please use esp_spp_enhanced_init [-Wdeprecated-declarations]
   73 |     if ((ret = esp_spp_init(ESP_SPP_MODE_CB)) != ESP_OK) {
      |     ^~
In file included from c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_spp.c:16:
C:\Users\Julien\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32/include/bt/host/bluedroid/api/include/api/esp_spp_api.h:269:11: note: declared here
  269 | esp_err_t esp_spp_init(esp_spp_mode_t mode) __attribute__((deprecated("Please use esp_spp_enhanced_init")));
      |           ^~~~~~~~~~~~
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_spp.c: In function 'ps3_spp_callback':
c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_spp.c:142:9: warning: 'esp_bt_dev_set_device_name' is deprecated: Please use esp_bt_gap_set_device_name or esp_ble_gap_set_device_name [-Wdeprecated-declarations]
  142 |         esp_bt_dev_set_device_name(PS3_DEVICE_NAME);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\Julien\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3_spp.c:15:
C:\Users\Julien\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32/include/bt/host/bluedroid/api/include/api/esp_bt_device.h:99:11: note: declared here
   99 | esp_err_t esp_bt_dev_set_device_name(const char *name) __attribute__((deprecated("Please use esp_bt_gap_set_device_name or esp_ble_gap_set_device_name")));
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1`
with this code `#include <Ps3Controller.h>

void setup()
{
    Serial.begin(115200);
    Ps3.begin("01:02:03:04:05:06");
    Serial.println("Ready.");
}

void loop()
{
  if (Ps3.isConnected()){
    Serial.println("Connected!");
  }

  delay(3000);
}`  

I  expect I can help to fix the issue

parvaizahmad, 2025-04-26:
you can use virtual controller: https://youtu.be/DiOjzlxyG8o

robotdelta, 2025-04-29:
I haven't tried it because for my project I needed to use a PS3 controller. I found a solution to my problem  using the old version 3.1.0 of the ESP32 cards library by Espressif systems. It seems to resolve all of my errors and compiling successfully.