jvpernis/esp32-ps3 #88: Compilation error:  PS3 Controller Host

CarGT88, 2025-04-13:
Hi jvpernis,

I am new to the subject and have no idea.

My goal is to use Arduino IDE (2.3.6) to upload a code to an ESP-32 dev board and couple it with a PS3 controller.
I followed the instructions and got stuck on the “compilation of the sketch”.

The output highlights the following lines in red:

"
_d:\---\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3.c: In function 'ps3SetBluetoothMacAddress':
d:\---\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3. c:287:5: error: implicit declaration of function 'esp_base_mac_addr_set' [-Wimplicit-function-declaration]
 287 | esp_base_mac_addr_set(base_mac);
 | ^~~~~~~~~~~~~~~~~~~~~_
"

and ends with 

"
_exit status 1
Compilation error: exit status 1_
"

Do you have an idea what I can do?

Best Regards
CarGT88

SebastianStoll, 2025-04-25:
Duplicate of #90

parvaizahmad, 2025-04-26:
you can use virtual controller: https://youtu.be/DiOjzlxyG8o

porthos62, 2025-04-27:
@SebastianStoll  - thanks for the fix. I've downgraded the ESP32 library as suggested and the sketch now compiles.