jvpernis/esp32-ps3 #90: PS3 and ESP32S3

AlanLL, 2025-04-21:
c:\Users\Alan_\Documents\Arduino\libraries\PS3_Controller_Host\src\ps3.c:287:5: error: implicit declaration of function 'esp_base_mac_addr_set' 
287 |     esp_base_mac_addr_set(base_mac);
I have used your controller on previous ESP32 boards and all have been compiled and work sucessfully, however I get the above error when using ESP32S3

SebastianStoll, 2025-04-25:
Had the same issue.

This was caused by an upgrade of the used Espressif ESP32 library in my Arduino IDE setup to version 3.2.0.

See this issue for more: https://github.com/espressif/arduino-esp32/issues/11262

It can be fixed by just downgrading to e.g. version 3.1.0 of the ESP32 library.

AlanLL, 2025-04-25:
Hello SebastianStoll ,

My controller is a PS3 and NOT PS4

Regards
Alan Lovett

See this issue for more: espressif/arduino-esp32#11262<https://github.com/espressif/arduino-esp32/issues/11262>

SebastianStoll, 2025-04-25:
Hej Alan,

I also have a PS3 controller. What is not working is the function in the ESP32 library. It is used the same way for a PS3 or PS4 controller. Just try to downgrade the ESP32 library to version 3.1.0 and it should compile again.

Best regards,
Sebastian

parvaizahmad, 2025-04-26:
you can try virtual controller, ref: https://youtu.be/DiOjzlxyG8o